berry-lang/berry #6: More English documents.

gztss, 2018-12-01:


MuresanVladMihail, 2019-02-27:
This would be very useful, especially the back-end part of the language!

gztss, 2019-02-27:
Yes, I am writing a Chinese manual now, but I don’t have enough ability to translate.

https://github.com/gztss/berry_doc

MuresanVladMihail, 2019-02-27:
Do it as you can and I'll help out!

gztss, 2019-02-28:
Ok, I will start translating.

dcurrie, 2020-04-17:
FYI, I opened .tex pages from the documentation repo in the Google Chrome browser, and Google Translate offered to convert Chinese (Simplified) to English. The first few lines of [the type_variable page](https://github.com/Skiars/berry_doc/blob/master/zh_cn/type_variable.tex) are:

> \ chapter { type and variable }
> 
> \ textbf { type } is an attribute of data, which defines the meaning of the data and the operations that can be performed on the data. Types can be divided into built-in types and user-defined types. Built-in types refer to some basic types built into the Berry language, and types that are not based on class definitions are called \ textbf { simple types }. The type defined based on the class is called \ textbf { class type }, some of the built-in types belong to the class type, and user-defined types are also class types.
> 
> \ section { built-in type }
> 
> \ subsection { simple type }
> 
> \ subsubsection { Nil }
> 
> The Nil type is the null type, which means that the object has an invalid value, and it can also be said that the object has no meaningful value. This is a very special type. Although we may say that a variable is \ texttt { nil }, the nil type actually has no value, so what we are talking about here is that the variable type is nil (instead of value).
> 
> The default value of a variable before assignment is \ texttt { nil }. This type can be used for logical operations. In logical operations, \ texttt { nil } is equivalent to \ texttt { false }.
> 
> \ subsubsection { integer type }
> 
> The integer type (integer) represents a signed integer, referred to as an integer. The number of bits of the integer represented by this type depends on the specific implementation. In 32-bit platforms, there are usually 32-bit signed integers. Integer is a type of arithmetic and supports all arithmetic operations. When using this type of integer range is to be noted, typically 32-bit signed integer that ranges from $ - 2147483648 $ to $ 2,147,483,647 $ between.
> 
> \ subsubsection { real number type }
> 
> Real type (real), to be precise floating point type. Real number types are usually implemented as single-precision floating-point numbers or double-precision floating-point numbers. The real number type is also an arithmetic type. Compared with the integer type, the real type has higher precision and a larger range of values, so this type is more suitable for mathematical calculations. It should be noted that the real number type is actually a floating-point number, so there is still a precision problem. For example, it is not recommended to compare two \ texttt { real } type values ​​for equality.
> 
> Integers and real numbers are usually converted to real numbers when participating in operations at the same time.
> 

This could be a quick way to get started on a translation.

Skiars, 2020-04-18:
It looks good, but the details need to be optimized.

I have been very busy recently, and I should start to update the document after a while.

Skiars, 2021-12-13:
I set up a new [repositories](https://github.com/berry-lang/berry-lang.github.io) and plan to put the documents in it:

hiperiondev, 2022-10-24:
It's this issue still relevant?

closed 2022-10-24 by s-hadinger