berry-lang/berry #15: A parser bug that cannot handle an error condition

gztss, 2019-02-26:
Parse the following code
``` ruby
a = 1
b = 0
a = b (b) = 1
```
We expect to get an error message stating the third line `(b) = 1` but the program will crash.

MuresanVladMihail, 2019-02-26:
This also seems to fail:
```
a = 1
b = 0
a = b = 1
```
At first I thought it's something related with the parentheses but it's not that (or only that).

gztss, 2019-02-26:
These problems are caused by the same bug. You can check the [`assign_expr`](https://github.com/gztss/berry/blob/da7369c2723650740b9bfe0f1c6bb70b095553d2/src/be_parser.c#L632) function.

Berry does not support continuous assignment, so it should be reported at the second operator `=` of `a = b = 1`. This problem may be due to an error handling of the operator `=`.

MuresanVladMihail, 2019-02-26:
My naive solution to fixing this problem is this: 
```C
if (next_token(parser).type == TokenId) {
			parser_error(parser, "Invalid assignment");
		}
```

gztss, 2019-02-26:
Please let me see some context.

MuresanVladMihail, 2019-02-26:
```C

static void assign_expr(bparser *parser)
{
    int line;
    bexpdesc e;
    init_exp(&e, ETVOID, 0);
    expr(parser, &e);
    line = parser->lexer.linenumber;
    if (next_token(parser).type == OptAssign) { /* '=' */
        bexpdesc e1;
        init_exp(&e1, ETVOID, 0);
        scan_next_token(parser);
		if (next_token(parser).type == TokenId) {
			parser_error(parser, "Invalid assignment");
		}
        expr(parser, &e1);
        check_var(parser, &e1);
        if (e.type == ETVOID) { /* new variable */
            new_var(parser, e.v.s, &e);
        }
        if (be_code_setvar(parser->finfo, &e, &e1)) {
            parser->lexer.linenumber = line;
            parser_error(parser,
                "try to assign constant expressions.");
        }
    } else { /* not assign expression */
        check_var(parser, &e);
    }
}
```

gztss, 2019-02-26:
There are some problems with this solution. For example, the following valid expressions will fail to compile:
``` ruby
a = 0
b = a # i think this will be misreported.
```

MuresanVladMihail, 2019-02-26:
Correct, I just discovered that and the fact that you cannot assign function to it, while I was doing some tests for it.

gztss, 2019-02-26:
Yes, so you have to do more verification.

MuresanVladMihail, 2019-02-26:
What do you think about this: 
```C

static void assign_expr(bparser *parser)
{
    int line;
    bexpdesc e;
    init_exp(&e, ETVOID, 0);
    expr(parser, &e);
    line = parser->lexer.linenumber;
    if (next_token(parser).type == OptAssign) { /* '=' */
        bexpdesc e1;
        init_exp(&e1, ETVOID, 0);
        scan_next_token(parser);
        expr(parser, &e1);
        check_var(parser, &e1);
        if (e.type == ETVOID) { /* new variable */
            new_var(parser, e.v.s, &e);
        }
        if (be_code_setvar(parser->finfo, &e, &e1)) {
            parser->lexer.linenumber = line;
            parser_error(parser,
                "try to assign constant expressions.");
        }
		if (next_token(parser).type == OptAssign) {
			parser_error(parser, "Invalid assignment!");
		}
    } else { /* not assign expression */
        check_var(parser, &e);
    }
}
```
Test case:
```js 
a = 1
b = 0
def v()
	print("hi")
end 

a = v() 
a = 1
b = 1;
a = b;
print (a); 
a = b = a;
```

gztss, 2019-02-26:
Please test this code:
``` ruby
= 1
```

MuresanVladMihail, 2019-02-26:
I was testing exactly that before seeing your comment!

MuresanVladMihail, 2019-02-26:
I think the best approach to it is to check if before token '=' we have a token TokenId

gztss, 2019-02-26:
Yes, but I think my plan is feasible.

MuresanVladMihail, 2019-02-26:
```C
static void assign_expr(bparser *parser)
{
	if (next_token(parser).type != TokenId) {
		parser_error(parser, "Expected token id!");
	}
	int line;
	bexpdesc e;
	init_exp(&e, ETVOID, 0);
	expr(parser, &e);
	line = parser->lexer.linenumber;
	//printf("%d\n", e.type);
	if (next_token(parser).type == OptAssign) { /* '=' */
		bexpdesc e1;
		init_exp(&e1, ETVOID, 0);
		scan_next_token(parser);
		expr(parser, &e1);
		check_var(parser, &e1);
		if (e.type == ETVOID) { /* new variable */
			new_var(parser, e.v.s, &e);
		}
		if (be_code_setvar(parser->finfo, &e, &e1)) {
			parser->lexer.linenumber = line;
			parser_error(parser,
					"try to assign constant expressions.");
		}
		if (next_token(parser).type == OptAssign) {
			parser_error(parser, "Invalid assignment!");
		}
	} else { /* not assign expression */
		check_var(parser, &e);
	}
}
```

I'm starting to think berry needs some unit testing :)

gztss, 2019-02-26:
Yes, I am not good at this. I did write a simple test framework

gztss, 2019-02-26:
This bug should be fixed, please see this commit [844eaf0](https://github.com/gztss/berry/commit/844eaf0e671f96dd6519f6c661ecb978fe6fda55).

MuresanVladMihail, 2019-02-26:
Works fine for me!

gztss, 2019-02-26:
Ok, I think we need to be certified to consider the test. Of course the documentation is also a big problem. My English is very poor.

closed 2019-02-27 by gztss