berry-lang/berry #17: Feature request: assert function

MuresanVladMihail, 2019-02-26:
I was trying to add a new function, assert, in baselib.c but soon I came across an impairment that that is: the lack of getting the current line and file from bvm structure.  
From what I understand reading the source code this is not possible. Please advise.

gztss, 2019-02-26:
The bvm structure is designed to be invisible to external apis, and the berry api interface operates on a virtual stack.

gztss, 2019-02-26:
Please wait a moment, I will write some tutorials for this.

gztss, 2019-02-26:
I made some simple instructions, please see this page: [API Example](https://github.com/gztss/berry/wiki/API-Example). If you need any help, please tell me.

MuresanVladMihail, 2019-02-27:
Thanks for the example but I already figured how to do that :) 
What I was trying to say is this: Right now there's no way of knowing the location where a "bvalue" has been created at (file and line number). I need this information in order to create an "assert" function (on the C side), eg:
```js 
def assert(cond) 
if (!cond)
  push_error(linenumber, filename)
end
end
The solution(?) I came with is to have this modified in "be_object.h":
```C
typedef struct {
#if BE_RUNTIME_DEBUG_INFO > 1
    uint16_t linenumber;
    uint16_t endpc;
#else
    int linenumber;
    int endpc;
#endif
} blineinfo;

typedef struct bvalue {
    union bvaldata v;
    int type;
    blineinfo lineinfo;
} bvalue;
```
And basically everytime you create a variable or modify the value of it change the the information accordingly. I would also add in structure blineinfo a const char *fname.

gztss, 2019-02-27:
I don't provide such an API, but you can do it.

Please refer to "be_debug.c", this file has examples of source code information at runtime.

MuresanVladMihail, 2019-02-27:
I will, thank you!

gztss, 2019-02-27:
Calling the C function `be_debug_error` will terminate the program and output a runtime error message, I think it should be what you want.

MuresanVladMihail, 2019-02-27:
Not quite, it doesn't report correctly the line number if it's called in the global space aka main function but works correctly if I call it into another function.
```js 
def a()
 def init()
 assert(1 > 2) #works fine! 
end 
end 
assert(1 > 2) # doesn't work fine, the line number is wrong
```

gztss, 2019-02-27:
You need trace call stack.

gztss, 2019-02-27:
Do you have any progress?

I think it would be easier to assert the implementation as a C function. Or, we can implement an error function straight throw, this time will naturally display the call information.

gztss, 2019-02-27:
You can commit the code to your repository and I can provide some advice.

MuresanVladMihail, 2019-02-27:
I implemented the assert function as a C function in be_baselib.c and using the blineinfo structure added in bvalue but sometimes I don't get the right line number on which the assert function is being called on the _berry_ side. I have to look more into it.

gztss, 2019-02-27:
You can run this code:
``` ruby
a = 0
print(a + 'str')
```
Then you will get the following runtime error message:
``` ruby
stack traceback:
        test.be:2: in function `main`
        [C]: in ?
```
Is this expected?

MuresanVladMihail, 2019-02-27:
I get the same output as you. I'll see if I can reuse some error handling logic from operator '+' and apply it to the assert function.

gztss, 2019-02-27:
If an error occurs while calling the function, the call stack will be printed, like:
``` ruby
def func(x)
    return x + 'str'
end
print(func(0))
```
error message:
```
test.be:2: error: unsupported operand type(s) for +: 'int' and 'string'
stack traceback:
        test.be:2: in function `func`
        test.be:4: in function `main`
        [C]: in ?
```

MuresanVladMihail, 2019-02-27:
If functions would take as a parameter in be_baselib.c the parser structure then error checking would be much easier to handle :)

gztss, 2019-02-27:
If you call the function [`be_debug_error`](https://github.com/gztss/berry/blob/36be542d7721283d0d7221d9af2eee4793810bd7/src/be_debug.c#L175), you will get the debug message.

gztss, 2019-02-27:
It will be easier, but it is not conducive to the realization of the external parser.

MuresanVladMihail, 2019-02-27:
```C
static int l_assert(bvm *vm)
{
    int pass = be_tobool(vm, 1);

    if (!pass) {
        be_debug_error(vm, BE_EXEC_ERROR, "Assert failed!");
    }
    be_return_nil(vm);
}
```

```js
class Base1
    var b
    def init(a)
        self.b = a * 3
    end
end
class A1 : Base1
    var c
    def init(c)
        self.c = c
        super(self).init(c)
    end
    def func(a, b)
        return 1 + a + b
    end
    def -*(other)
    end
end

assert(1 > 2) # line 20 in editor 

obj = A1(2)
print(obj.func(5, 6))
print("exiting from class.be") 
```

Output: 
native error: Assert failed!
	<native>: in native function
	test/class.be:24: in function `main`
	[C]: in ?

gztss, 2019-02-27:
Nice!

MuresanVladMihail, 2019-02-27:
The line number in editor does not match the one in stack trace

gztss, 2019-02-27:
Emmm...

gztss, 2019-02-27:
It seems that if the error is thrown in the native function, there will be problems with the stack information.

MuresanVladMihail, 2019-02-27:
Yes!

gztss, 2019-02-27:
I think the function [`sourceinfo`](https://github.com/gztss/berry/blob/36be542d7721283d0d7221d9af2eee4793810bd7/src/be_debug.c#L100) may have a problem.

MuresanVladMihail, 2019-02-27:
Isn't it related to https://github.com/gztss/berry/blob/36be542d7721283d0d7221d9af2eee4793810bd7/src/be_debug.c#L107 ?

gztss, 2019-02-27:
check this https://github.com/gztss/berry/blob/36be542d7721283d0d7221d9af2eee4793810bd7/src/be_vm.c#L470

gztss, 2019-02-27:
Please wait a moment, I will confirm.

gztss, 2019-02-27:
Please review the new submission and this issue should be resolved.

Since the IP (instruction pointer) is not pushed onto the stack when the native function is called,
There is a problem with the calculation of the PC. This is not a VM bug, but a performance consideration. Now I have written the `patch_native` function to correct the IP value when I need to print debugging information.

MuresanVladMihail, 2019-02-27:
Now it works perfectly! Well done! I was looking at IP and PC this whole time, glad you figure it out!

MuresanVladMihail, 2019-02-27:
Do you have a more in depth explication why IP is not pushed onto the stack when calling the native function (native function == main function,right?)?

closed 2019-02-27 by MuresanVladMihail

gztss, 2019-02-27:
Native functions are functions implemented in C, such as `print` and `assert`.

Because there is no bytecode when calling a native function, IP is not used and there is no need to push it onto the stack. If the Berry function is called after this, the IP will be pushed onto the stack and restored after this function returns, so there is no problem.

MuresanVladMihail, 2019-02-27:
I understand! Thanks!

gztss, 2019-02-27:
If you can, please do further testing.

MuresanVladMihail, 2019-02-27:
I will!