berry-lang/berry #20: test.be: 20: error: 'list' undeclared (first use in this function)

rofl0r, 2019-02-26:
if i change `list` on line 20 to `lout`, i get a segfault:
```
Program received signal SIGSEGV, Segmentation fault.
0x000000000040aecc in i_getglobal (vm=0x7ffff7d6dee8, ins=1208221695)
    at src/be_vm.c:276
276         *v = vm->global[idx];
(gdb) p idx
$1 = 262143
#0  0x000000000040aecc in i_getglobal (vm=0x7ffff7d6dee8, ins=1208221695)
    at src/be_vm.c:276
#1  0x000000000040e30b in vm_exec (vm=0x7ffff7d6dee8) at src/be_vm.c:715
#2  0x000000000040e701 in do_closure (vm=0x7ffff7d6dee8, reg=0x624378, argc=0)
    at src/be_vm.c:771
#3  0x000000000040e964 in be_dofunc (vm=0x7ffff7d6dee8, v=0x624378, argc=0)
    at src/be_vm.c:804
#4  0x0000000000409ed6 in m_pcall (vm=0x7ffff7d6dee8, data=0x7fffffffe9e0)
    at src/be_exec.c:96
#5  0x0000000000409d4f in be_execprotected (vm=0x7ffff7d6dee8, 
    f=0x409ea1 <m_pcall>, data=0x7fffffffe9e0) at src/be_exec.c:62
#6  0x0000000000409f5f in be_protectedcall (vm=0x7ffff7d6dee8, v=0x624378, 
    argc=0) at src/be_exec.c:108
#7  0x000000000041767b in be_pcall (vm=0x7ffff7d6dee8, argc=0)
    at src/be_api.c:763
#8  0x000000000041979b in dofile (vm=0x7ffff7d6dee8) at default/berry.c:103
#9  0x0000000000419992 in analysis_args (vm=0x7ffff7d6dee8)
    at default/berry.c:167
#10 0x0000000000419a52 in main (argc=2, argv=0x7fffffffeb58)
    at default/berry.c:187

```

gztss, 2019-02-26:
This bug was fixed in commit [36be542](https://github.com/gztss/berry/commit/36be542d7721283d0d7221d9af2eee4793810bd7).

rofl0r, 2019-02-26:
works for both issues :+1:

closed 2019-02-26 by rofl0r