berry-lang/berry #21: Test Case

gztss, 2019-02-27:
I think it is necessary to use an automated test to verify the interpreter. As an exploration, here is a simple automated test framework:

https://github.com/gztss/berry_testcase

MuresanVladMihail, 2019-03-05:
My idea is something similar to how you validate if a scheme language is valid to the standard implementations and that is: 
Have a file named `test_language.be` in which there will be all known operations supported by _berry_ and everytime before commiting a new change this `test_language.be` must be ran in order to see if the new changes didn't break the language.

gztss, 2019-03-05:
This is a good idea, I don't have much idea for this goal.

Old test tools are still available, maybe we can modify them on this basis.

closed 2021-12-13 by Skiars