berry-lang/berry #22: Check for return type in methods like: tostring, tobool

MuresanVladMihail, 2019-02-28:
I saw that berry has some predefined functions like: tostring and tobool (among others). 
Right now _berry_ allows for code like:

```ruby 
class base 
	def init()
	end 
	def tostring()
                # returning void is also considered okay 
		return 2 #results in segment fault for me 
	end 
end 
b = base();
print (b)
```

From my point of view this is not entirely correct so I'm suggesting a solution: doing type checking for predefined functions like those mentioned above. 

Below https://github.com/gztss/berry/blob/master/src/be_strlib.c#L103 add this if statement: 
```C
if (top->type != BE_STRING) {
            be_debug_error(vm, 1, "expected return type string in function 'tostring");
        }
```

Something similar can be applied for the rest of predefined functions. Let me know what you think.

MuresanVladMihail, 2019-02-28:
After looking at how _berry_ handles the function **tobool**  I don't think any more checks have to be made, looking forward for your opinion.

gztss, 2019-02-28:
Yes, this is very necessary. However, I suggest you put the declaration of `be_debug_error` in the berry.h file and give it a better name.

Then you can define a macro as an API, like:
``` c
#define be_throw_msg(vm, msg)  be_debug_error((vm), BE_EXEC_ERROR, (msg))
```

https://github.com/gztss/berry/blob/9c193ea9c8a9d08527b682d3a7a91969911f88f2/src/berry.h#L30

Obviously, the native functions should use `BE_EXEC_ERROR`.

MuresanVladMihail, 2019-02-28:
I totally agree!

MuresanVladMihail, 2019-02-28:
Since Github was bought by Microsoft I don't use it for hosting anymore, I uploaded the changes here if you want to see: http://tildesquad.ddns.net:3001/murii/berry/commit/f4f95fedc3ad380633f312d8889d1e7bdfb8b07f

Off topic: what does this comment mean:

> print("fib:", fib(35)) # minimum stack size: 78!!

> 
Are you referring to the depth of the recursion before the stack is full?

gztss, 2019-02-28:
A long time ago, *berry* needed to specify the stack size (number of registers), which means that running `fib(35)` would require a stack of 78 registers. Now *berry* will automatically expand the stack, so this information is useless.

MuresanVladMihail, 2019-02-28:
Oh, awesome!

gztss, 2019-02-28:
I saw your changes, I suggest:

* Change the name of the function `be_debug_error` (all-replace)
* Put the declaration of the function `be_debug_error` at the bottom of the *berry.h* file.
* Remove the declaration of the function `be_debug_error` in *be_debug.h*.
* Do not use the `extern` modifier (unless you add the `extern` modifier to all functions).

MuresanVladMihail, 2019-02-28:
Ok, I did it.

gztss, 2019-02-28:
I think using *print* is not accurate. This function does:
* Generate an error message and convert it to a string
* Push the error message onto the top of the stack.
* Throw an exception and terminate execution.

Do you have a better plan?

MuresanVladMihail, 2019-02-28:
Not sure to be honest.

gztss, 2019-02-28:
Please check these commits and may be useful to you:

https://github.com/gztss/berry/commit/f668cdd7ae4fce517db8ba6f80aabcc2b482155a

MuresanVladMihail, 2019-02-28:
Why did you drop the **bint** typedef? Maybe on some platforms it can be defined as having another value like: short or long.

gztss, 2019-03-01:
I changed the argument of the function `be_pushint` from `bint` to `int`, but used `bint` internally.

If we determine that `bint` should be used in APIs, I will change the argument of the function `be_pushint` to `bint` and change the return value of `be_toint` to `bint`.

closed 2019-05-11 by Skiars