berry-lang/berry #23: Read and use passed arguments to berry

MuresanVladMihail, 2019-03-01:
There should be a way of making this possible: `./berry folder/script.be argv1 argv2 hi` and from `script.be` I should be able to do this: `print (argv[1])`.

gztss, 2019-03-01:
This is a very important feature. I will develop it.

One option is to push command line arguments as strings onto the stack. Just like the transfer of function arguments.

MuresanVladMihail, 2019-03-01:
Tried making a `list` but I can't seem to figure out how to use the API correctly :\

gztss, 2019-03-02:
Making a `list` instance can be cumbersome, so I wrote a simple tutorial: [Instantiate a list object in a native function](https://github.com/gztss/berry/wiki/FFI-Example#instantiate-a-list-object-in-a-native-function)

MuresanVladMihail, 2019-03-03:
I'll have to take a closer look at the whole VM but I think will do just fine. Please let me know what you think :)
http://tildesquad.ddns.net:3001/murii/berry/commit/184bdf302626457c91674d7e784709eaa72a0d49

gztss, 2019-03-03:
I think you have done a lot of work, so I feel that reading the VM code completely helps you.

I have an immature idea of using `arg` as a argument to the `main` function instead of providing an `arg` function.

Now I will think more about improving the documentation, which seems to be a big problem.

MuresanVladMihail, 2019-03-03:
Improving the documentation especially the VM part would be extremely helpful since that part is the one more complicated compared the the rest. 
Overall, I really like your project and I'm quite amazed by how fast it performs compared to Ruby, Python and Lua!

gztss, 2019-03-03:
Is it fast? 😂

Then, my computer is broken and I can't do a lot of work in a week.

MuresanVladMihail, 2019-03-03:
Yeah, I find it very fast :1st_place_medal:

gztss, 2019-03-03:
My main goal is memory optimization.

Skiars, 2021-02-11:
Berry's default interpreter already supports this feature.

closed 2021-02-11 by Skiars