berry-lang/berry #24: Missing bitwise operators

MuresanVladMihail, 2019-03-03:


MuresanVladMihail, 2019-03-03:
So far I added the bitwise operator '&' but I have to make more tests because I found some problems.
http://tildesquad.ddns.net:3001/murii/berry/commit/a52cefa2cef57d5e681cc510610fffb4c2857fbb

gztss, 2019-03-04:
Very good, this is what I always wanted to do. It is recommended not to add the `BIT` prefix to the opcode, for example, the bit and operation are written as `OP_AND`, because the bitwise operators are not confusing. You should find that logical operations do not have corresponding opcodes (they use conditional jumps).

Regarding the opcode, the relevant opcode name for the X86 assembly is:
* `NOT`: flip bits
* `AND`: bit and
* `OR`: bit or
* `XOR`: bit XOR
* `SHR`: unsigned right shift
* `SHL`: unsigned left shift

MuresanVladMihail, 2019-03-04:
Thanks, I'll modify !

MuresanVladMihail, 2019-03-04:
I also added the last bitwise operator, "~", but I have a question regarding it. 
```rb 
a = ~4
```
Generated bytecode: 
```
   0: LDINT	R0	-5
```

The expected output is indeed "-5". I'd like to know if you think the outputted bytecode is correct.  You can find the changes on my git. Thanks!

gztss, 2019-03-04:
This seems to be a compile-time evaluation, but I almost forgot the specific process. 😂

I haven't found the '~' submission record yet.

MuresanVladMihail, 2019-03-05:
You can see the logic for `~` here: http://tildesquad.ddns.net:3001/murii/berry/src/branch/master/src/be_code.c#L457

gztss, 2019-03-05:
You need to test the types of expressions, most of the time they are not constants but variables. You can refer to the implementation of the negative operator `-`.

MuresanVladMihail, 2019-03-05:
Fixed it!

MuresanVladMihail, 2019-03-05:
To be 100% honest I didn't really understand why I had to do what I did but I'll look more into it. If you have some tips I'm all ears :D

gztss, 2019-03-05:
First, only integer numeric literals can be flipped at compile time.

For the place you comment, you are actually working on an expression or variable. Therefore, you need to get the register that stores the expression value:
``` c
int src = exp2anyreg(finfo, e);
```
This line of code processes the expression and gets the location where the value is stored. Make it either a local variable or a temporary register. If it is a temporary register, we can directly use this register to save the result of the bit flip operation. Otherwise we have to allocate a new register:
``` c
int dst = e->type == ETREG ? src : be_code_allocregs(finfo, 1);
```
Finally we generate bytecode based on this information:
``` c
codeABC(finfo, OP_NOT, dst, src, 0);
```

gztss, 2019-03-11:
Are you planning to submit a PR for these jobs?

closed 2019-04-13 by Skiars