berry-lang/berry #25: Add error when stack limit reached end

MuresanVladMihail, 2019-03-03:
```rb
def aaaa() 
	return aaaa() 
end 
aaaa();
```
This code results in segment fault. I think this is due to hitting the limit of the stack size. If that's the case I think _berry_ needs to issue an error.

gztss, 2019-03-04:
No, I don't have this problem here:
```
test.be:2: error: stack overflow (maxim stack size: BE_STACK_TOTAL_MAX)
stack traceback:
    test.be:2: in function `aaaa`
    test.be:2: in function `aaaa`
    ...
    [C]: in ?
```

MuresanVladMihail, 2019-03-04:
In that case I will have a closer look at this problem.

gztss, 2019-03-04:
Well, maybe this is a very hidden question.

MuresanVladMihail, 2019-03-04:
Tested the same build on Ubuntu 18.06 and I got the same error like you showed above but on my main work machine I get this:
`

[GCC 4.8.5 20150623 (Red Hat 4.8.5-36)] on Linux (default)

Starting program: /home/vmuresan/Programming/berry/./berry test/test2.be
Hello from test2.be!
ARG 0 is:  /home/vmuresan/Programming/berry/./berry

Program received signal SIGSEGV, Segmentation fault.
0x000000000040b681 in premark_stack (vm=0x61c018) at src/be_gc.c:330
330	            gc_setgray(var_togc(v));
Missing separate debuginfos, use: debuginfo-install glibc-2.17-260.el7_6.3.x86_64 ncurses-libs-5.9-14.20130511.el7_4.x86_64 readline-6.2-10.el7.x86_64
(gdb) bt
#0  0x000000000040b681 in premark_stack (vm=0x61c018) at src/be_gc.c:330
#1  be_gc_collect (vm=vm@entry=0x61c018) at src/be_gc.c:438
#2  0x000000000040b8ce in be_gc_auto (vm=vm@entry=0x61c018) at src/be_gc.c:430
#3  0x000000000040b91d in be_gc_newstr (vm=vm@entry=0x61c018, 
    size=size@entry=17, islong=islong@entry=0) at src/be_gc.c:86
#4  0x000000000040948f in createstrobj (vm=vm@entry=0x61c018, len=len@entry=0, 
    islong=islong@entry=0) at src/be_string.c:103
#5  0x00000000004095ac in newshortstr (len=0, str=0x412bc7 "", vm=0x61c018)
    at src/be_string.c:142
#6  be_newstrn (vm=vm@entry=0x61c018, str=str@entry=0x412bc7 "", 
    len=len@entry=0) at src/be_string.c:175
#7  0x0000000000406578 in pushstr (len=0, s=0x412bc7 "", vm=0x61c018)
    at src/be_strlib.c:126
#8  be_pushvfstr (vm=0x61c018, 
    format=0x414cb0 "%s error: %s\nstack traceback:\n", 
    arg=arg@entry=0x7fffffffdbc8) at src/be_strlib.c:144
#9  0x0000000000403f77 in be_pushfstring (vm=vm@entry=0x61c018, 
    format=format@entry=0x414cb0 "%s error: %s\nstack traceback:\n")
    at src/be_api.c:269
#10 0x000000000040aed2 in addinfo (
    msg=0x414868 "stack overflow (maxim stack size: BE_STACK_TOTAL_MAX)", 
    vm=vm@entry=0x61c018) at src/be_debug.c:182
#11 be_pusherror (vm=vm@entry=0x61c018, errcode=errcode@entry=2, 
---Type <return> to continue, or q <return> to quit---

`

MuresanVladMihail, 2019-03-04:
I'll also test the code on my Macbook (latest OS version)

MuresanVladMihail, 2019-03-04:
This is my solution, please look over it and tell me if you find it correct because I'm not sure if it really is but at least it works. 
```C
static void addinfo(bvm *vm, const char *msg)
{
    be_gc_setpause(vm, 0);
    bcallframe *cf = vm->cf;
    if (var_isclosure(cf->func)) {
        char buf[100];
        be_pushfstring(vm, "%s error: %s\nstack traceback:\n",
                       sourceinfo(vm, buf, -1), msg);
        } else {
        patch_native(vm, -1);
        be_pushfstring(vm, "native error: %s\n", msg);
    }
    tracestack(vm);
    be_gc_setpause(vm, 1);
}
```
Maybe you can give me a little explication? I'd appreciate it!

MuresanVladMihail, 2019-03-04:
From what I understand _berry_ pushes new strings and the garbage collector is being called and it fails at marking the gray area but I don't know why(yet at least).

gztss, 2019-03-05:
I found the reason using valgrind debugging:
```
berry git:(master)  valgrind --tool=memcheck --leak-check=full ./berry test1.be
==552== Memcheck, a memory error detector
==552== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==552== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==552== Command: ./berry test1.be
==552==
==552== error calling PR_SET_PTRACER, vgdb might block
==552== Conditional jump or move depends on uninitialised value(s)
==552==    at 0x11E0A5: premark_stack (be_gc.c:329)
==552==    by 0x11E58D: be_gc_collect (be_gc.c:438)
==552==    by 0x11E565: be_gc_auto (be_gc.c:430)
==552==    by 0x11D350: be_gc_newstr (be_gc.c:86)
==552==    by 0x11A5D8: createstrobj (be_string.c:103)
==552==    by 0x11A7C8: newshortstr (be_string.c:142)
==552==    by 0x11A938: be_newstrn (be_string.c:175)
==552==    by 0x11B17F: pushstr (be_strlib.c:123)
==552==    by 0x11B27C: be_pushvfstr (be_strlib.c:141)
==552==    by 0x10F0EB: be_pushfstring (be_api.c:269)
==552==    by 0x11D0EE: addinfo (be_debug.c:179)
==552==    by 0x11D15B: be_pusherror (be_debug.c:189)
==552==
```
The reason is that the stack is pushed for it before the string is created. This makes the GC possible to set gray for an invalid pointer. The modified commit is https://github.com/gztss/berry/commit/000a3410fbee1cab20b93e6cc06a05f6f0497fd1, it should solve the problem.

It's wise to turn off the GC in error handling. However, we should try our best to ensure that there is no problem when the GC is not turned off.

MuresanVladMihail, 2019-03-05:
This fixes the problem!

closed 2019-03-05 by MuresanVladMihail