berry-lang/berry #26: Add warning about lexer keywords and their order

MuresanVladMihail, 2019-03-05:
```C
/* IMPORTANT: This must follow the enum found in be_lexer.h !!! */
static const char* const kwords_tab[] = {
        "NONE", "EOS", "ID", "INT", "REAL", "STR",
        "=", "+", "-", "*", "/", "%", "<", "<=",
        "==", "!=", ">", ">=", "..", "<<", ">>", "^", "|", "&", "&&",
        "||", "!",  "~",
        "(", ")", "[", "]", "{", "}", ".", ",", ";",
        ":", "if", "elif", "else", "while", "for",
        "def", "end", "class", "break", "continue",
        "return", "true", "false", "nil", "var", "do",
        "import", "as", "require",
};
```
At first I didn't realize I must follow the pattern but then my tests started to fail and after an investigation I figured it out.

gztss, 2019-03-05:
Nice, welcome to PR.

MuresanVladMihail, 2019-03-05:
https://github.com/gztss/berry/pull/27

closed 2019-03-05 by MuresanVladMihail