berry-lang/berry #29: Add 'require' keyword for using other classes

MuresanVladMihail, 2019-03-06:
Berry needs a keyword which enables it to use more than just one file programs. I added a such keyword here: http://tildesquad.ddns.net:3001/murii/berry/src/branch/master/src/be_parser.c#L1103
Maybe you'll find this interesting

gztss, 2019-03-06:
I think this feature should be implemented using a native function. On the other hand, we should expand the `import'keyword.

We can use native functions to run other files, such as the `parser` function, for example:

*other.be*:
``` ruby
print('this is a external source file')
```
run:
``` ruby
file = open('other.be', 'r')
fun = parser(file.read())
fun() # 'this is a external source file'
```

MuresanVladMihail, 2019-03-06:
I like what you're suggesting. I'll see what I can do about it!

MuresanVladMihail, 2019-03-06:
Don't you think this code: 
```rb
file = open('other.be', 'r')
fun = compile(file.read())
```

can be replaced with this code:
```rb
fun = compile('other.be')
```
I'm asking because I'd like to reduce the C calls as much as possible to save performance.

MuresanVladMihail, 2019-03-06:
```C
char* be_readfile(FILE *fp, size_t *len)
{
    char *buffer;
    fseek(fp,0L,SEEK_END);
    *len = ftell(fp);
    buffer = malloc(*len);
    fseek(fp, 0L, SEEK_SET);
    if (buffer) {
        *len = fread(buffer, 1, *len, fp);
    }
    fclose(fp);
    return buffer;
}
```

```C
static void import_stmt(bparser *parser)
{
    /*
     * When this is set to true we are importing another
     * file from the disk not module!
     */
    bbool requires = bfalse;
    bstring *name;
    bexpdesc m, v;
    /* 'import' ID ['as' ID] */
    /* 'import' String  */
    scan_next_token(parser); /* skip 'import' */
    name = next_token(parser).u.s;
    if (next_token(parser).type == TokenString) {
        requires = btrue;
        init_exp(&m, ETSTRING, 0);
        m.v.s = name;
    } else {
        requires = bfalse;
        match_token(parser, TokenId); /* match and skip ID */
        init_exp(&m, ETSTRING, 0);
        m.v.s = name;
    }
    if (next_token(parser).type == KeyAs) {
        scan_next_token(parser); /* skip 'as' */
        name = next_token(parser).u.s;
        match_token(parser, TokenId);  /* match and skip ID */
    }
    if (!requires) {
        new_var(parser, name, &v);
        be_code_import(parser->finfo, &m, &v);
    } else {
        const char *cstr_fname = cast(char*, cast(bstring*, name) + 1);
        FILE *fp = fopen(cstr_fname, "r");
        if (!fp) {
            char buf[128];
            sprintf(buf, "Couldn't open file named: %s\n", cstr_fname);
            parser_error(parser, buf);
            return;
        }
        size_t len;
        char *text = be_readfile(fp, &len);

        /* save these two so we can get back to them later */
        const char *save_fname = parser->lexer.fname;
        const char *save_cursor = parser->lexer.cursor;

        size_t text_len = parser->lexer.text_len;

        be_lexer_set_source(&parser->lexer, cstr_fname, text, len);
        scan_next_token(parser);
        stmtlist(parser);

        be_lexer_set_source(&parser->lexer, save_fname, save_cursor, text_len);
        scan_next_token(parser);
        stmtlist(parser);

        free(text);
    }
}
```
Usage: 

```rb 
import "test/class.be" # I'll look into how I can make the 'as' keyword to work here
#or 
import math as m 
```

gztss, 2019-03-06:
What you said makes sense. One of the meanings of runtime compilation is interaction, so you must be able to compile strings. I will design the `compile` function like this:

``` ruby
compile('some code', ['string']) # compile string
compile('file name', 'file')     # compile file
```

MuresanVladMihail, 2019-03-06:
Ok, this function will be placed in `be_baselib.c` ?

gztss, 2019-03-06:
Yes, I am already doing it.

Then I suggest that you only do one thing with one function.

gztss, 2019-03-06:
You can refer to this submission, and I will not do the work of compiling the file.

I am going to write a set of interfaces to read the source text stream.

MuresanVladMihail, 2019-03-06:
Function `l_compile' needs more checks. Example: 
```rb
compile("test/list.be") -> segment fault 
compile("list.be") -> does nothing
```
I know that this is not the way the function needs to be used but others might not ;)

gztss, 2019-03-06:
OK, I will.

gztss, 2019-03-06:
The problem is solved https://github.com/gztss/berry/commit/fbaf6064a66e0c3e1700b8d8c40c19d315345142.

gztss, 2019-03-07:
I modified the lexer and implemented it.
`be_loadfile` function.
Now you can execute `compile('test.be', 'file')`.

MuresanVladMihail, 2019-03-08:
I'll test

closed 2021-02-11 by Skiars