berry-lang/berry #36: Compile to bytecode, to load and execute at a later time

stunndard, 2019-10-06:
Is it possible to compile to a bytecode file (similar to .luac file) ?
So that bytecode file can be taken and executed at another environment, where only minimal berry runtime is present, and no compiler available.  That would be very useful on small and embedded systems, where there's no need to have source files in plaintext, and compiler is also not needed.

Looking at be_api.c file didn't help much.

Isn't `code_dump` what I'm looking for?

Skiars, 2019-10-06:
I have never implemented outputting bytecode to a file. This is a good suggestion, but some problems must be solved first.

The `code_dump` only displays the bytecode of the function, it should not be what you want.

The bytecode file needs to agree on the data format. Another problem is the index of the global variable (Berry uses the index to access the global variable, so what about the native global object?)

stunndard, 2019-10-06:
I'm currently implementing a bytecode save/loading feature for a similar language that was supposed to compile and execute the bytecode in one go, with no ability to save bytecode to a file, so I had to invent the bytecode format myself.

What I basically did, is walk all structures and objects for compiled program in memory, and output everything serialized to a file. I faced some problems where, for example, the compiled bytecode contained raw pointers to initialized global and module-level variables. So when that bytecode is loaded, I had to manually recreate all variables, structures, etc in memory from the file, then fix up the addresses in bytecode to point to newly created objects and variables.

When dumping vars and objects, I had to add one extra field, called `address`. It contains an address of variable/object at the moment after compilation. 

When loading that bytecode, first I recreate all vars and objects, then check compiled bytecode ASM for instructions that are dealing with var addresses, so when I find an address that points to a needed variable, I just patch the bytecode with the address of newly created variable or object. Unbelievable, but it works, for a language and VM that never supposed to be divided into separate compiler and runtime components, and never supposed to have a bytecode.

Skiars, 2019-10-06:
In my limited experience, your thinking should be correct. Berry's bytecode does not contain any pointers (only indexes exist). The handling of native global variables and object pointers should be a big challenge.

Correcting the pointer should be necessary. For Berry, the index values of global objects may also be processed (each runtime may have a different native object order). Global objects can also be accessed by name, so a name-index mapping table may be needed in the bytecode file.

I think this process is similar to compiling, except that bytecode files are used to generate bytecode.

stunndard, 2019-10-08:
I don't think you need file offsets at all.

I mentioned pointers because some other language/VM is using raw pointers in its bytecode directly, like this:
```
  PUSHGLOBALVAR 0x12364646
```
Where `0x12364646` is the memory address of the initialized object. If such bytecode is moved to another machine or just reloaded, the address of object would change, so bytecode would crash. Berry is not using pointers, but indexes, and that makes things easier.

After compile, imagine you have an object that's referenced by an index in the "executable" bytecode.
You just serialize and save that object, together with all other objects that are present in the VM state after compilation, before or after the executable bytecode, but in the same file.
You can also have global or native variables, you save them in the same way and put in the bytecode file.

```
struct Object {
  value: int
  name: string
  ...etc
  index: int
}

struct Func1 {
  name: string
  code..
  ..etc
  index: int
```

When loading that bytecode file, first you load and de-serialize all objects and native variables etc and put them in memory. Their indexes will be also restored that way. Then you can load the actual "executable" bytecode (which has references to objects by index), and everything should work.

bytecode example:
```
  PUSHLOCALVAR Object1.index
  PUSHINT 0
  CALLFUNC Func1.index
 ....
  MOV R0, Object1.index
```

No matter where in memory global objects are, the bytecode still can reference them by index.
I have interest in Berry because of its small footprint and also because it's a register VM, which is faster than stack VM.  Bytecode feature would be nice.

Skiars, 2019-10-10:
You gave me some inspiration, and I realized that I really don't have to worry about indexing. We only need to serialize all the objects and arrange them by index, while the bytecode remains the same.

For Berry, the most complex object might be a function (prototype) that contains some object references and other details. Berry's global variables should not need to be serialized. Any global variables should not be initialized at compile time, and the VM state will not be modified at compile time. These points should simplify the work.

Another point is that compile time does not construct structures such as List and Map. The types of values generated at compile time include:

* Nil
* Boolean
* Number (integer and real)
* String
* Function (prototype)
* Class

Function and class are complex objects that may contain other values. We also need to consider how strings are stored. As for other types, it's simple, so you only need to store the value.

Skiars, 2019-10-12:
After the above discussion, I thought of the structure of a bytecode file, which is organized as follows:

``` lua
-------------------------------------------------------------------------------
--                             description
--
--         a double dash('--') start a line comment.
-- a: n    means that the size of entity 'a' is 'n' bytes.
-- 'a: .n  means that the size of entity 'a' is 'n' bits.
-- a:
--      b  means that the entity 'a' is realized by the 'b' structure.
-- a:
--      b
--      c  'b' and 'c' are arranged in a compact order (one byte alignment).
-- [a]     means that structure 'a' can be repeated 0 to any times.
-- [a](b)  means that structure 'a' can be repeated 'b' times.
-- a | b   means that the entity can be implemented by structure 'a' or 'b'.
-- a -> b  is equivalent to 'a:
--                               b'.
--         only the first entity is a file entity (the root).
-------------------------------------------------------------------------------

bytecode_file:  -- little endian
    header
    main_function

header: 8
    magic_number: 2 -- 0xbecd (berry code)
    version: 1      -- update with file structure definition
    verify_code: 1  -- user defined
    integer_size: .1
    float_size: .1
    -- reserved space

main_function -> function

function:
    information:
        function_name:
            string
        argc: 1  -- arguments count
        stack_size: 1
    bytecode:
        code_size: 4
        code_array: code_size * 4
    constant_table:
        constant_count: 4
        [constant_value]{constant_count}
    debug_info:
        -- reserved

constant_value:
    type: 1  -- type of value
    nil | boolean | integer | float | string | function | class

string:
    string_size: 4
    byte_array: string_size

class:
    class_name:
        string
    member_count: 4   -- number of member variables
    method_count: 4   -- number of method
    member_index_table -> [
            string    -- member name
            int          -- member variable index
        ](member_count)
    method_table -> [
            string    -- method name
            function  -- method function body
        ](method_count)

nil: 1
boolean: 1
integer: 4 or 8
float: 4 or 8
```

Skiars, 2019-10-12:
I have already started working:

https://github.com/Skiars/berry/tree/bytecode
https://github.com/Skiars/berry/blob/bytecode/tools/grammar/berry.bytecode

Skiars, 2019-10-18:
Now all the work has been merged into the master branch.

stunndard, 2019-10-28:
All seems working, thank you for this big work.
I will make more bytecode tests.

Small suggestion: is it possible to change loading bytecode from:
`berry -b file.bec`

to

`berry file.bec`

so it has a uniform call to execute a text script or compiled bytecode script

Skiars, 2019-10-29:
It should be possible to identify valid bytecode files by header information.

closed 2019-12-03 by Skiars