berry-lang/berry #37: compiler error

sealindx, 2019-10-10:
When I compiled Betty, reported an error:

lindx@lindx-PC:~/local/berry$ make
[Make] map_build
[Prebuild] generate resources
terminate called after throwing an instance of 'std::regex_error'
  what():  Invalid range in bracket expression.
Makefile:59: recipe for target 'generate/be_const_strtab.h' failed

sealindx, 2019-10-10:
I switched a machine (ubuntu 14.04LTS) to compile Berry, but also error

`lucky@ubuntu:~/Documents/berry$ make
[Make] map_build
map_build.o: In function `map_build::parse_attr(std::string const&)':
map_build.cpp:(.text+0x3cfc): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator(__gnu_cxx::__normal_iterator<char const*, std::string>, __gnu_cxx::__normal_iterator<char const*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, std::bitset<11ul>)'
map_build.cpp:(.text+0x3d09): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator()'
map_build.cpp:(.text+0x3d46): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator!=(std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> > const&)'
map_build.cpp:(.text+0x3d58): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x3daf): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x3efb): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator++()'
map_build.o: In function `map_build::parse_body(std::string const&)':
map_build.cpp:(.text+0x4494): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator(__gnu_cxx::__normal_iterator<char const*, std::string>, __gnu_cxx::__normal_iterator<char const*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, std::bitset<11ul>)'
map_build.cpp:(.text+0x44a1): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator()'
map_build.cpp:(.text+0x44e1): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator!=(std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> > const&)'
map_build.cpp:(.text+0x44f6): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x458a): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x46e1): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator++()'
map_build.o: In function `map_build::parse_block(std::string const&)':
map_build.cpp:(.text+0x493a): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator(__gnu_cxx::__normal_iterator<char const*, std::string>, __gnu_cxx::__normal_iterator<char const*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, std::bitset<11ul>)'
map_build.cpp:(.text+0x4947): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator()'
map_build.cpp:(.text+0x4971): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator!=(std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> > const&)'
map_build.cpp:(.text+0x4a0c): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x4a86): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x4b01): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x4c16): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
map_build.cpp:(.text+0x4e7b): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator++()'
main.o: In function `builder::info_block(std::string const&)':
main.cpp:(.text+0xbdc): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator(__gnu_cxx::__normal_iterator<char const*, std::string>, __gnu_cxx::__normal_iterator<char const*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, std::bitset<11ul>)'
main.cpp:(.text+0xbe9): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_iterator()'
[... cut ...] egex_traits<char> > const&)'
str_map.cpp:(.text+0x1a8): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator->()'
str_map.cpp:(.text+0x2c3): undefined reference to `std::regex_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::operator++()'
collect2: error: ld returned 1 exit status
make[1]: *** [map_build] Error 1
make: *** [tools/map_build/map_build] Error 2
`

Skiars, 2019-10-10:
Please tell me your compiler version. `std::regex` is a C++11 feature, so the compiler must support C++11.

sealindx, 2019-10-10:
lucky@ubuntu:~/Documents/berry$ g++ --version
g++ (Ubuntu 4.8.4-2ubuntu1~14.04.4) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.

Skiars, 2019-10-10:
It is recommended that you update the compiler and library. I am not sure if this version of g++ supports c++11 full regular expression syntax.

sealindx, 2019-10-10:
I upgraded g++ to 7.4.0, it is ok
thank you

sealindx, 2019-10-10:
that's a bad thing. some versions of Linux don't compile

Skiars, 2019-10-10:
Yes, but it is not easy to find alternatives to `std::regex` to refactor these features.
I rarely test the code on older versions of the compiler, so I didn't notice this problem. I should modify these codes to improve compatibility if necessary.

closed 2019-12-03 by Skiars