berry-lang/berry #38: Questions about be_map.c

sealindx, 2019-10-12:
When I read Berry's source code, I found a puzzle in be_map.c 

```c
static int eqnode(bmapnode *node, bvalue *key, uint32_t hash)
{
    bmapkey *k = key(node);
    if (hashcode(key(node)) == hash && (int)k->type == key->type) {
        switch (key->type) {
        case BE_NIL:
            return 0;
        case BE_BOOL:
            return key->v.b == k->v.b;
        case BE_INT:
            return key->v.b == k->v.b;
        case BE_REAL:
            return key->v.r == k->v.r;
        case BE_STRING:
            return be_eqstr(key->v.s, k->v.s);
        default:
            return key->v.p == k->v.p;
        }
    }
    return 0;
}
```

when case BE_INT,  it is should be comparing int type, not bool. I don't know if I'm right ?

Skiars, 2019-10-12:
Yes, I am wrong here.

sealindx, 2019-10-12:
我看了下你的信息，发现你应该是个中国人，能加个qq，交个朋友吗

closed 2019-10-12 by sealindx

Skiars, 2019-10-12:
你私信我的邮箱吧 skiars@qq.com