berry-lang/berry #45: How to compile on windows 10?

Fruin, 2020-05-19:
Hello,

I use mingw + msys2 to compile on Windows 10 but it gives the following makefile error:

[Prebuild] generate resources
tools/map_build/map_build.exe: tools/map_build/map_build.exe: cannot execute binary file
make: *** [Makefile:66: generate/be_const_strtab.h] Error 126

Skiars, 2020-05-19:
I tried the latest msys2 (gcc 9.3.0), it does not compile directly, but *map_build.exe* has no problem.

I need to check until the gcc version you are using, use the following command:
``` bash
gcc -v
```

You need to check after executing `make` (even if it is wrong):
``` bash
ls tools/map_build
```

You need to check if *map_build.exe* exists.

Skiars, 2020-05-19:
I am not familiar with msys. If you want to use Berry on windows, I recommend you use [MinGW-w64](https://sourceforge.net/projects/mingw-w64). You can use **mingw32-make** to build in *cmd* or *PowerShell* (maybe you need to set environment variables).

Skiars, 2020-05-19:
This [commit 9758f7a](https://github.com/Skiars/berry/commit/9758f7a514efda817dfb425ee66b96a1e0f688ee) already supports cygwin and msys2.

HAHEU, 2020-05-19:
Hello @Skiars,

I'm using Windows 10 too, I tried compiling with mingw and msys but it gives the following error:

> [Prebuild] generate resources
> terminate called after throwing an instance of 'std::regex_error'
>   what():  Invalid range in bracket expression.
> Makefile:65: recipe for target 'generate/be_const_strtab.h' failed
> mingw32-make: *** [generate/be_const_strtab.h] Error 3

There is map_build.exe in the tools/map_build file.

Can you create a project file so that it can be compiled more easily for Visual Studio? (.Sln extension)

Skiars, 2020-05-20:
Visual Studio project: [berry_vs.zip](https://github.com/Skiars/berry/files/4654311/berry_vs.zip)

Can you tell me your compiler version again? As far as I know, the error above is caused by incomplete compiler support for regular expressions (earlier versions of gcc have this problem).

HAHEU, 2020-05-20:
gcc (MinGW.org GCC-6.3.0-1) 6.3.0

Skiars, 2020-05-20:
OK.

I plan to remove the dependency on regex in *map_build* in the future.

HAHEU, 2020-05-20:
Okay thank you.