berry-lang/berry #47: Suggestion for the next version

HAHEU, 2020-05-26:
Hi @Skiars ,
I will have a suggestion for the next release. These:

* Variable names must be single and unique.

```
> var a = 10
> var a = 'hi'
> print(a)
hi
```
Enum, const variable, Recursion function. etc. should be added.

We should be able to use the code we wrote in the CLI interface. (For Windows)
`./berry myfirstcode.be`

We should be able to customize the command line:

`Console.BackgroundColor = Color.red` (C#) like...

That's all I want to add for now. Thank you

HAHEU, 2020-05-26:
Hi @Skiars 

Can I replace Lexer? for example:

```
if (n)              --> instead
   ...

если (n)
   ...
```

Skiars, 2020-05-27:
### Customize Lexer

You can customize lexer. To avoid unnecessary problems, first, modify this macro in *berry_conf.h*:

``` c
#define BE_USE_PRECOMPILED_OBJECT       0
```

Then modify the kwords_tab array in *be_lexer.c*.

### New features

* The variable declaration behavior you describe is easy to do and worth considering
* Berry supports recursive functions
* Enum, const variable. etc. Currently I have no idea.
* Custom terminal interface? This is not the core function of the language, so you can consider implementing the library for a specific platform.

HAHEU, 2020-05-27:
hi @Skiars ,

#define BE_USE_PRECOMPILED_OBJECT       0 --> I changed it to 0

### static const char* const kwords_tab[] = {
    "NONE", "EOS", "ID", "INT", "REAL", "STR",
    "=", "+=","-=", "*=", "/=", "%=", "&=", "|=",
    "^=", "<<=", ">>=", "+", "-", "*", "/", "%",
    "<", "<=", "==", "!=", ">", ">=", "&", "|",
    "^", "<<", ">>", "..", "&&", "||", "!", "~",
    "(", ")", "[", "]", "{", "}", ".", ",", ";",
    ":", "?", "->", "если", "elif", "else", "while",

I changed kword_tab like this but it gives the following error:
`syntax_error: stdin:1: stray '\167' in program
`

Skiars, 2020-05-28:
You have to look here (in *be_lexer.c*):

``` c
static btokentype lexer_next(blexer *lexer)
{
    for (;;) {
        switch (lgetc(lexer)) {
        // ...
        default:
            if (is_letter(lgetc(lexer))) {
                return scan_identifier(lexer);;
            }
            be_lexerror(lexer, be_pushfstring(lexer->vm,
                "stray '\\%d' in program", (unsigned char)lgetc(lexer)));
            return TokenNone; /* error */
        }
    }
}
```

closed 2020-05-29 by HAHEU