berry-lang/berry #50: I need help with Lexer

Nuowo, 2020-06-24:
Hi, I am trying to translate the statements into my language, but it gives me "syntax_error: stdin: 1: stray '\ 167' s program" error.

MY LEXER:

"NONE", "EOS", "ID", "INT", "REAL", "STR",
    "=", "+=","-=", "*=", "/=", "%=", "&=", "|=",
    "^=", "<<=", ">>=", "+", "-", "*", "/", "%",
    "<", "<=", "==", "!=", ">", ">=", "&", "|",
    "^", "<<", ">>", "..", "&&", "||", "!", "~",
    "(", ")", "[", "]", "{", "}", ".", ",", ";",
   ":", "?", "->", "wenn", "swenn", "sonst", "während",
    "zum", "funktion", "ende", "klasse", "brechen", "fortsetzen",
    "rückkehr", "wahr", "falsch", "nil", "var", "machen",
    "importieren", "wie", "versuchen", "außer", "erziehen"

The expression I use is:

```
klasse person
   var name, age
   funktion init(name, age)
     self.name = name
     self.age = age
   ende

  funktion tostring()
     rückkehr 'name: ' + str(self.name) + ', age: ' + str(self.age)
   ende
ende
```

I think the problem stems from this method. This method checks the English letters. How can I sort the letters that are not in the English alphabet?

```
static int is_letter(int c)
{
    return (c >= 'A' && c <= 'Z') || (c >= 'a' && c <= 'z') || (c == '_');
}
```

Skiars, 2020-06-25:
You can look at this link first:  [https://github.com/Skiars/berry/issues/47#issuecomment-635056504](https://github.com/Skiars/berry/issues/47#issuecomment-635056504)

Perhaps, you need to lift the English alphabet restrictions here:

``` c
static int is_letter(int c)
{
    return (c >= 'A' && c <= 'Z') || (c >= 'a' && c <= 'z') || (c == '_') || (c >= 128);
}
```

If you want to support non-ASCII characters better, you must improve lexer's support for unicode.

Nuowo, 2020-06-25:
What can I do about the Unicode lexer?

Can you help me? Where should I start?

Skiars, 2020-06-25:
I have no idea yet, but this should not be necessary.

The above function `is_letter` should also work.

Nuowo, 2020-06-25:
Oh okay, Thanks.

closed 2020-06-25 by Nuowo