berry-lang/berry #55: Two questions about split and trim.

enty8080, 2020-07-01:
@Skiars 

**1.** How to split using char like in python?

**python:**

```
line = "Hello, world!"
print(line.split(" ")) # ['Hello,', 'world!']
```

**2.** How to trim `\n` from `input()`?

```
>>> a = input('> ')
> hi
>>> print(a)
hi

>>> 
```

Sorry for this stupid question, but I can't find any english documentation about Berry.

Skiars, 2020-07-01:
I haven't implemented perfect string support yet.

There is a `split` function in the `string` library, which only supports splitting using integer indexes:

https://github.com/Skiars/berry/blob/4507c8c7b58b71ea1698700a2e921055c6ac8bf2/src/be_strlib.c#L463

``` python
import string
print(string.split("01234", 2)) # ['01', '234']
print(string.split("01234", -2)) # ['012', '34']
print(string.split("Hello, world!\n", -1)[0]) # "Hello, world"
```

No string splitting based on matching has been implemented yet.

In addition, the string is not an instance, it does not support member operation (like `a.b`). In the future, I might consider providing member operators for strings (after all, this is very useful).

enty8080, 2020-07-01:
Why `input()` adds `\n`. Why Berry can’t automatically trim `\n` from `input()` result like in python?

Skiars, 2020-07-02:
You can modify this feature, just change the implementation of the `input` function

enty8080, 2020-07-02:
@Skiars Input fix - #57

Skiars, 2020-07-02:
Got it.

Skiars, 2020-07-03:
I implemented some functions:

``` ruby
string.find(s1, s2, begin=0, end=length(s2))
string.count(s1, s2, begin=0, end=length(s2))
string.split(s, position)
string.split(s1, s2, count=string.count(s1, s2))
```

closed 2021-02-11 by Skiars