berry-lang/berry #59: Question about delay

enty8080, 2020-07-02:
Is there any delay function in Berry like in python `time.sleep()`?

Skiars, 2020-07-02:
Not implemented yet.

enty8080, 2020-07-02:
@Skiars maybe I made this function, but it works like this:

```
>>> import time
>>> time.sleep(1) # ok
>>> time.sleep(0.5) # ok
>>> time.sleep() # infinity
```

enty8080, 2020-07-02:
If user puts nothing to `time.sleep()` then what is into `vm`?

Skiars, 2020-07-03:
`be_toreal` returns a double value:

``` c
breal be_toreal(bvm *vm, int index);
```

Usually time_t is defined like this:

``` c
typedef long time_t;
```

finally:

``` c
time_t second = be_toreal(vm, 1);  // => (long)0.5 == 0
```

enty8080, 2020-07-03:
I think I should use:

```c
double second = be_toreal(vm, 1)
```

enty8080, 2020-07-03:
I print second after `float second = be_toreal(vm, 1)` and it returned this

```
>>> import time
>>> time.sleep(0.1)
0.100000
```

In code

```c
sleep(second); // sleep(0.100000) = 0?
```

enty8080, 2020-07-03:
Debug:

```
debug.c:4:8: warning: implicit conversion from 'double' to 'unsigned int' changes
      value from 0.1 to 0 [-Wliteral-conversion]
        sleep(0.100000);
        ~~~~~ ^~~~~~~~
1 warning generated.
```

enty8080, 2020-07-03:
I wrote my own sleep function and it works with float numbers.

enty8080, 2020-07-03:
@Skiars #60

Skiars, 2020-07-03:
I don't think the delay of using loops is really sleep, it will consume a lot of CPU without suspending thread.

You can use `usleep` in Linux:

``` c
#include <unistd.h>
int usleep (useconds_t usec);

breal sec = 0.1;
usleep((useconds_t)(sec * 1000000)); // seconds to microseconds
```

There is no similar API for Windows.

Skiars, 2020-07-03:
If an API is platform dependent, please consider using *default/be_port.c* and *src/be_sys.h* for compatibility.

enty8080, 2020-07-03:
**Solotion:**

```c
#if defined(_WIN32)
    #include <windows.h>
    #define delay(x) Sleep((x*1000))
#else 
    #include <unistd.h>
#endif 

static int m_sleep(bvm *vm)
{
    if (vm != NULL) {
        double milisecs = be_toreal(vm, 1);
        unsigned int secs = milisecs * 1000;
        delay(secs);
        be_return_nil(vm);
    } else {
        be_return_nil(vm);
    }
}
```

Skiars, 2020-07-03:
Yes. I think you can refer to the implementation of CPython.

enty8080, 2020-07-03:
```
src/be_timelib.c: In function ‘m_sleep’:
src/be_timelib.c:61:9: error: implicit declaration of function ‘usleep’; did you mean ‘sleep’? [-Wimplicit-function-declaration]
   61 |         usleep(secs);
      |         ^~~~~~
      |         sleep
make: *** [Makefile:58: src/be_timelib.o] Error 1
```

?

enty8080, 2020-07-03:
Replace `-std=c99` with `-std=gnu99` in Makefile to use unistd.h with usleep

enty8080, 2020-07-03:
It works!

Skiars, 2020-07-03:
Don't use gnu99, just use c99.

CPython sleep implementation:

https://github.com/python/cpython/blob/91e1bc18bd467a13bceb62e16fbc435b33381c82/Modules/timemodule.c#L1888

CPython uses `select` function to realize the `sleep` of Unix-like systems.

enty8080, 2020-07-03:
@Skiars berry is cross-platform, time.sleep() should work in windows too, not only in Unix-like systems.

Skiars, 2020-07-03:
Sorry, I am not complete, the following link actually has sleep implementation under Windows. I just pointed out a sleep implementation that does not depend on gnu99 in Linux.

https://github.com/python/cpython/blob/91e1bc18bd467a13bceb62e16fbc435b33381c82/Modules/timemodule.c#L1888

The implementation in this link is definitely cross-platform 😉.

enty8080, 2020-07-03:
@Skiars Why we should not use `gnu99` instead of `c99`?

Skiars, 2020-07-03:
GNU99 comes with some GNU extensions, and currently I want to use the ISO C99 standard as much as possible. There is no special reason.

closed 2021-02-11 by Skiars