berry-lang/berry #62: try-except does not work.

enty8080, 2020-07-06:
```ruby
try
    undeclaratedvariable
except .. as e, v
    print("error")
end
```

This is not work, it should return `error` but it returns `'undeclaratedvariable' undeclared`. Why?

Skiars, 2020-07-06:
This is a compilation error, try-except can only catch runtime errors (when the compilation has not been completed, we do not know that there is a try-cache block).

The following compilation errors can be catch:

```
try
    compile('undeclaratedvariable')
except .. as e, v
    print("error")
end
```

At this time the compilation error in `compile` will be catch.

enty8080, 2020-07-06:
@Skiars one more question about `do` keyword. What `do` does?
Is there any difference between:
```
do
    print("Hello, world!")
end
```
and just
```
print("Hello, world!")
```
?

Skiars, 2020-07-07:
`do` - `end` only provides a scope level, similar to C or JS's `{ ... }`

closed 2021-02-11 by Skiars