berry-lang/berry #63: Few questions.

enty8080, 2020-07-09:
**1.** What `__iterator__` does?

**2.** Why `isinstance` does not work?

```
>>> type(a)
instance
>>> isinstance(a)
false
```

**3.** What `classof` does and `issubclass` does?

**I know this is stupid question, but there is no english docs :(**

Skiars, 2020-07-10:
This is a short manual: [short_doc.pdf](https://github.com/Skiars/berry/files/4900425/short_doc.pdf)

enty8080, 2020-07-12:
@Skiars Does berry has function `from math import *` to not use `math.pow(5, 2)`, just `pow(5, 2)`?

Skiars, 2020-07-13:
There is currently no such syntax.

enty8080, 2020-07-15:
@Skiars,  I have a question about your REPL, when I trying to use tab in it, it does not works and shows me directory contents:

```
> (tab)
.idea    myfiles samples
>> (tab)
.idea    myfiles samples
```

Skiars, 2020-07-17:
Can you elaborate? I don't understand.

enty8080, 2020-07-23:
@Skiars everything is ok. This is a problem in my OS :)

Skiars, 2020-07-24:
Okay, it looks strange.

closed 2020-08-05 by Skiars