berry-lang/berry #65: Callback function

engelant, 2020-08-04:
Is it possible to take a function off the stack inside an native function and put it back on the stack at a later time for execution to implement a callback?

Skiars, 2020-08-04:
It can be used like this (use `be_register`):

``` c
static int cb_index;
// berry native function
int reg_callback(bvm *vm)
{
    be_unregister(vm, cb_index);
    cb_index = be_register(vm, 1); // args[1] is the callback function
    be_return_nil(vm);
}

// ...
// c runtime
extern bvm *g_vm;
void such_as_interrupt(void)
{
    be_getregister(g_vm, cb_index);
    be_call(m_vm, 0);
}
```

You have to pay attention to some thread issues. These attachments provide some references: [be_qttimer.zip](https://github.com/Skiars/berry/files/5020509/be_qttimer.zip).

engelant, 2020-08-04:
I would like to use this mechanism (maybe even in conjunction with an language grammar extension consisting of await) to do asyncrounous execution in an uC environment.

```
an_callback_function("command",
    def(result)
        #do something with the result when it's available
        print(result)
    end,
    def(error)
        #do something in an error case
        print(error)
    end
)
```

This expresses the same
```
try
    result = await an_async_function("command")
    #do something with the result when it's available
    print(result)
except 'promise_rejected' as error
    #do something in an error case
    print(error)
```
Providing await a_native_async_function does a
`raise 'promise_rejected' error`

I figured since try/catch is already available and callback functions are easy to implement, this would add a nice touch i.m.h.o.

Skiars, 2020-08-05:
The asynchronous environment also has to deal with issues such as thread synchronization, which should still be a problem at present.

engelant, 2020-08-05:
Could you elaborate a little what you mean by thread syncronization?

Skiars, 2020-08-05:
Assuming that there is a global VM, you need to call Berry callbacks in interrupts or other threads, and the main interpretation thread may be executing scripts at this time. At this time, a certain mechanism is needed to ensure thread safety: only one thread uses the VM at the same time.

engelant, 2020-08-05:
Yes, that would be an issue.
On the other hand, are you familiar with the asyncio/workers concept, which is [implemented for nodejs](https://blog.logrocket.com/node-js-multithreading-what-are-worker-threads-and-why-do-they-matter-48ab102f8b10/).

They avoid the thread related issues by:
- not preemptivie interrupting a function
- moving the threads/workers parallelism to the OS level (may even be in a future step)

I understand this being a very fundamental descision in terms of architecture, but I find it quiet modern and like to use it.
And it's relatively easy to implement.
Just my 2 cents.

Skiars, 2020-08-05:
I am also considering threading/async.

The worker should be similar to running a new VM (it can be thought of as a thread).

I want a safe asynchronous callback mechanism, but I don't know how to implement it (I don't know the low-level implementation of nodejs and whether to use a lot of locks).

Skiars, 2020-08-05:
As far as I know, js maintains an event queue to implement asynchronous operations.

engelant, 2020-08-05:
Correct me if I'm wrong, but an implementation like this (I'm not very familiary with that subject) would do the trick.
Also this allows coroutines but doesn't touch threads, a vm function/callback is executed completely.
```c
msg_fifo = create_threadsafe_fifo();
main() {
    bvm *vm = be_vm_new();
    for(;;) {
        select(FD_SIZE, *msg_fifo, NULL, NULL, NULL);
        //take the first value from fifo, and prepare the vm
        be_getregister(g_vm, firstof(msg_fifo));
        be_call(vm, 0);
    }
    be_vm_delete(vm);
}
```
For an interactive shell one would have to fork the process, add the readline stuff and make it send/queue it's input via the msg_fifo to the main blocking event loop.

This would actually already be great and also this is something nodejs used up to version 12?
I a future step one could think of the requirements (or shamelessly copy the architecture of nodejs) to establish a worker pool.

With your first reply I actually already can implement said routine, but would have to use this anonymous callback code style, which I don't adore. Also a class Promise would be handy.

If I'm not missing anything this is quiet straight forward in terms of implementation and I want to find out, If you want your language to take this direction, since this idealy would mean altering the ebnf.

Skiars, 2020-08-05:
This is indeed a direct callback scheme, but `async`/`await` like js seems more elegant (this seems to be borrowed from C#).

The low-level of js `async` also uses a work queue, and I think it can be implemented as a function as a compromise:

```
o = async(def () ... end) # return a Promise object 
print(await(o))
```

engelant, 2020-08-05:
Since we're talking about compromise here, is the reason for compromising

- lack of time
- uncertenty of which solution to choose

I actually was searching quiet a while for a script language/interpreter, which:

- is lightweight and uC embeddable (RAM and ROM)
- offers modern language features
- has a nice tradeoff between size/featureset
- bytecode and script execution
  - public/standard libraries could be loded from bytecode while user specific scripts can be stored as text

Your project is great in term of all the points, I just really wish for async/await, as it makes coding so much cleaner.

Maybe I should close this discussion and reopen it as a feature request?

Skiars, 2020-08-06:
I just think the support of `async`/`await` can be divided into two steps:

1. First implement async/await with library functions;
2. Implement related syntactic sugar.

We should open a new issue.

Skiars, 2020-08-06:
Should we provide a library like python's asyncio?

engelant, 2020-08-06:
There is actually another issue, that came up related to the memory management of lists.
Since the promises get resolved at some point, the callback function saved on the registry should be deleted after execution.
But since we rely on the index and the list is basically a c array, the memory usage will just grow to infinity.

As for the 2 steps, I agree.

Regarding the python interface, I'm not quiet familiar with it.
What I know is the [Promise API from JavaScript](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise). I guess It's pretty much the same.

Skiars, 2020-08-06:
The registry is a memory pool. If the number of callbacks does not increase all the time, the registry will not increase.

The time complexity of registration and unregistration is O(1).

engelant, 2020-08-06:
As far as I'm aware of the callback architecture, there will be lots of promises, most of them anonymous and called once then deleted. Here's an example, but same goes for setTimout() and stuff like that:
```
do
    def cbf()
        print("Second callback")
        cbf2_test(cbf)
    end
    return cbf2_test(cbf)
end
```
with
```c
    if (cb2_index > 0) {
        be_getregister(g_vm, cb2_index);
        be_unregister(g_vm, cb2_index);
        cb2_index = 0;
        be_call(g_vm, 0);
    }
```

I was playing around with the idea of implementing a map with integer keys and a max value, to represent a list object, where valid items (0 ... n-1) which are not stored are returned as Nil.
Allthough, complexity goes dependent on the algorithms to O(log N) for access. As you propably can guess this is how node does it.

I also had a look at the code, and came to the conclusion that a different list interface might be required. Especially for iterating trough, to avoid the O(log N) for each access.

Skiars, 2020-08-06:
The insertion, deletion and query of the registry are all O(1):

https://github.com/Skiars/berry/blob/8a4189e61a9bfe17a705fa252b6466762bcd7b41/src/be_list.c#L176

The registry does not increase the capacity until the slot is used up.

You can look at this: [Fast Efficient Fixed-Size Memory Pool](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiarZOixYbrAhUpyYsBHSsXAdcQFjALegQIAhAB&url=https%3A%2F%2Fwww.thinkmind.org%2Fdownload.php%3Farticleid%3Dcomputation_tools_2012_1_10_80006&usg=AOvVaw2kKpQA4k8Y4aBq5KpD5Gtf)

engelant, 2020-08-06:
I got this, but if you remove some element, you get holes. And new values are always posted at the end.

Skiars, 2020-08-06:
No, you have to notice that we actually have a free linked list instead of appending to the end of the array.

You will find that this is almost a perfect allocator 😀.

engelant, 2020-08-07:
So it turns out I had an error in my code, leading to the list memmory runaway.
The callback index now behaves as expected, so that's for that.

closed 2020-08-07 by engelant