berry-lang/berry #72: Custom modules and Yield/Wait support, for embedding in Tasmota ESP32

s-hadinger, 2021-02-02:
First let me say that Berry is fantastic. I'm considering using Berry as a scripting language for Tasmota (https://github.com/arendst/Tasmota) running on ESP32. It's compact enough, powerful and easy to embed in existing code.

ESP8266 is too short in memory, but ESP32 makes a perfect target for the Berry language. I did a quick & dirty embeddeding Berry in Tasmota with no issue and it compiles fine. The documentation is very clear and easy to follow. Calling Berry functions from Tasmota, and the reverse works fine.

Now here are my questions:

1. Native modules: I tried to register native module used by `import` at runtime but I have the feeling that it's currently not supported except for the pre-included libraries. Do you plan to allow custom libs?

2. Yield/Wait: it would be fantastic to be able to `yield()` from within Berry code to give back control to the host code, and resume execution later. I understand that it's probably not straightforward, but this would make a great way to do async-style code that would not block the host code.

s-hadinger, 2021-02-03:
Thinking deeper into `yield`/`wait`, I believe it could be implemented using closures and sending back an array with first element being the number of ms to wait for, and the second the continuation function.

Ideally the code would look like:
```
def test()
  print('before')
  print('before2')

  yield            # of wait(0)

  print('after')
  print('after2')

end
```

And it would be actually compiled as:
```
def test()
  print('before')
  print('before2')

  return [0, def()  # added

  print('after')
  print('after2')

  end ]             # added

end
```

Example using REPL:
```
> cont = test()
before
before2
> cont
[0, <function: 0x7ffb90605350>]
> cont[1]()
after
after2
>
```

s-hadinger, 2021-02-03:
Scratch that. This is over-simplistic and wouldn't work for recursion and loops.

Looking at `vm_exec()`, maybe it would imply adding a new opcode in the bytecode, to exit `vm_exec()` and freeze the `vm` state so that it can be continued later.

Skiars, 2021-02-04:
For native modules, you can refer to [be_timelib.c](https://github.com/Skiars/berry/blob/master/src/be_timelib.c) and [be_modtab.c](https://github.com/Skiars/berry/blob/master/default/be_modtab.c). You need to provide be_modtab.c in your project.

Berry supports native module instances generated during compilation (special comments are processed by the *coc* tool to generate code). E.g:

``` cpp
/* @const_object_info_begin
module time (scope: global, depend: BE_USE_TIME_MODULE) {
    time, func(m_time)
    dump, func(m_dump)
    clock, func(m_clock)
}
@const_object_info_end */
```

You can check the operation of coc in the [Makefile](https://github.com/Skiars/berry/blob/master/Makefile).

s-hadinger, 2021-02-04:
Thanks for you response. I think I found a way to implement yield/wait outside of Berry using setjmp/longjmp.

More concretely, I allocate a separate stack for the berry vm when calling `be_pcall()`, then I used crossed setjmp/longjmp to switch between the host code and the berry code. I did a quick and dirty proof of concept and it works fine on ESP8266.

[image]

It's a little tricky from the caller side, but tweaking jmp_buf structure allows me to change the stack pointer without requiring assembly code (although the location of the stack is processor dependent). From Berry side, I simply need to add `yield` and `wait` native function, and it works with the vm as-is.

Skiars, 2021-02-05:
I have heard of coroutines implemented using setjmp/longjmp. For this, you have to be very careful to ensure the validity of setjmp/longjmp.

s-hadinger, 2021-02-05:
Agreed. But it looks like it's working fine. Setting the new stack is CPU dependent since it requires to change the `jmp_buf` opaque structure, and the change of stack requires the communication between both threads to go through global variables. That's not pretty, but I didn't find any easy way around this (yet).

closed 2021-02-05 by s-hadinger

s-hadinger, 2021-02-19:
FYI I will finally not go that route. I have added a timer system and it's simpler to just set a timer to trigger a closure in the future.