berry-lang/berry #74: How to iterate on map keys?

s-hadinger, 2021-02-10:
I didn't find any way to iterate on map keys, `iter()` actually returns a iterator for values, not keys:

```
> m={"a":1,"b":2,"c":3}
> for i: m print(i) end
1
2
3
```

Did I miss something?

s-hadinger, 2021-02-10:
I think I found the problem, changed `be_maplib.c` line 159:

From

```
    var_setstr(vm->top, &next->value);
```

To

```
    var_setobj(vm->top, next->key.type, next->key.v.p);
```

Now I get:
```
> m={"a":1,"b":2,"c":3} for i: m print(i) end
a
b
c
```

Do you want a PR?

Skiars, 2021-02-11:
I do not intend to allow the default iterator to support (key, value) iteration, because the efficiency may not be good. But we can consider returning a key-value pair iterator through a certain method of `map`. Like:

```
for i : { "a": "b", "c": "d" }.pairs()
    print(i) # ("a", "b"), ("c", "d")
end
```

s-hadinger, 2021-02-11:
Thanks for your answer. I was only asking for an iterator on keys, not necessarily (key,value) pairs. Or maybe we can add `map.iterkeys()` to have such iterator on keys.

Skiars, 2021-02-11:
Ok. We should be able to use it like this:

```
for i : m.keys()
    print(i, m[i])
```

I think this might be faster than `map.pairs()` (This will construct iteration objects many times, but Berry VM has no object cache yet).

s-hadinger, 2021-02-11:
Agreed :)

closed 2021-02-11 by s-hadinger