berry-lang/berry #78: Is it possible to precompile Berry code in Flash?

s-hadinger, 2021-02-19:
I understand how to have precompiled modules with native functions. However in Tasmota, out modules are partly made of native functions, and partly made of Berry code.
I have implemented a function to save precompiled bytecode in fs, and realized that `compile()` is able to load bytecode. However the bytecode is still in RAM.

Is there a way to have precompiled Berry code fully in Flash?

s-hadinger, 2021-02-27:
I looked deeper in the code and I'm refining my request. I would like actually to push into flash a Class and its code, taken that I don't call any global beyond the built-on Map() and List(), and no closure.

This seems feasible but it requires additional code to export the function structure (code and constants), and feed it into the coc compiler.

One other option would be to re-write the coc compiler in Berry, but that's certainly more work, and maybe not in your plans.

Skiars, 2021-03-03:
Now, the code must still be loaded into RAM for execution (even if we have a bytecode file). Maybe constants and codes can be solidify to Flash, but the pointers must be valid, and some codes of the Berry interpreter may need to be modified.

Maybe we can use some kind of tool to process bytecode files. As for the Berry interpreter, it is necessary to ensure that this solidified code is not GC.

s-hadinger, 2021-03-15:
I've been digging more in the Berry code. This is a masterpiece! Wow.

I'm working on a prototype of solidifying a `proto` that could be used by a native function at runtime. It's half-way from having a fully solidified function, but at least it saves to flash: bytecode, constants and hopefully the bproto struct.

The limitations would be:
- no upvals
- no globals (except builtins)
- no sub-proto

I'm struggling right now to solidify arbitrary strings and will need to add stuff to the coc parser.

Skiars, 2021-03-20:
Great, do you have a link to the code repository?

s-hadinger, 2021-03-22:
Not yet, I had to make a pause to complete the integration of Berry in Tasmota. This is now done. I will try further in the following weeks.

s-hadinger, 2021-03-22:
I had a first success. I compiled the following function:
```
def f(a,b) print("hello") return 1 end
```

I dumped it into a static `bclosure` structure as follows:
```
static const bvalue f_ktab[2] = {
  { { .s=(bstring*) &be_const_str_hello }, BE_STRING},
  { { .i=0x1 }, BE_INT},
};

static const uint32_t f_code[4] = {
  0x6008000F,  //  0000  GETGBL	R2	G15
  0x580C0000,  //  0001  LDCONST	R3	K0
  0x7C080200,  //  0002  CALL	R2	1
  0x80060200,  //  0003  RET	1	R257
};

static const bproto f_proto = {
  NULL,     // bgcobject *next
  8,       // type
  GC_CONST,        // marked
  4,       // nstack
  0,       // nupvals
  2,       // argc
  4,       // codesize
  2,       // nconst
  0,       // nproto
  NULL,     // bgcobject *gray
  NULL,     // bupvaldesc *upvals
  (bvalue*) &f_ktab,     // ktab
  NULL,     // bproto **ptab
  (binstruction*) &f_code,     // code
  (bstring*) &be_const_str_f,       // name
  (bstring*) &be_const_str_input,       // source
#if BE_DEBUG_RUNTIME_INFO /* debug information */
  NULL,     // lineinfo
  0,        // nlineinfo
#endif
#if BE_DEBUG_VAR_INFO
  NULL,     // varinfo
  0,        // nvarinfo
#endif
};

const bclosure f_closure = {
  NULL,     // bgcobject *next
  36,       // type
  GC_CONST,        // marked
  0,       // nupvals
  NULL,     // bgcobject *gray
  (bproto*) &f_proto,     // proto
  { NULL }     // upvals
};
```

Then I added:
```
BERRY_API void be_pushclosure(bvm *vm, void *cl)
{
    bvalue *reg = be_incrtop(vm);
    bclosure * closure = (bclosure*) cl;
    var_setclosure(reg, closure);
}
```

And finally declared the function as global:

```
    be_pushclosure(vm, (void*) &f_closure);
    be_setglobal(vm, "f");
    be_pop(vm, 1);
```

It worked which is reassuring. Lots of polishing to do now. Keeping the structures as `GC_CONST` and outside of the GC chain seems to prevent the gc from going crazy.

s-hadinger, 2021-03-23:
Here is a first working verion: https://github.com/s-hadinger/berry/tree/solidify_1
It's still early work, and part of conversion is quick and dirty.

```
import solidify
solidify.dump(<function>)
```

It prints C code that is then copy/paste in `be_testlib.c`

Constraints to solidifying function:
- no upvals
- no sub-protos
- no globals (except built-ins)
- no string literals longer than 254 characters (this could be added later)

Changes I needed to apply:

Patch BE_STRING: allow short strings to have an const value outside of the string pool. I.e. a short string (below 255 chars) with a null hash, is considered as a `bcstring` in flash. I patched `be_eqstr` and `be_strhash` to allow it. I hope I didn't forget anything else

Extend class definition to accept a second section for berry closures, patch `class_init`

``` 
    static const bnfuncinfo members[] = {
        { "dummy", NULL },

        { NULL, (bntvfunc) BE_CLOSURE }, /* mark section for berry closures */
        { "f", (bntvfunc) &f_closure },
        { "event", (bntvfunc) &event_closure },
        
        { NULL, NULL }
    };
```

The result looks like this (a bit ugly but efficient):

Original code:
```
def f(a,b) print(-1234567890123456) print(2.0/0.0333333333333333) return "Hello" end
import solidify solidify.dump(f)
```

```
/********************************************************************
** Solidified function: f
********************************************************************/

be_define_local_const_str(f_str_name, "f", 0, 0, 1, 0);
be_define_local_const_str(f_str_source, "stdin", 0, 0, 5, 0);
be_define_local_const_str(f_str_3, "Hello", 0, 0, 5, 0);

static const bvalue f_ktab[4] = {
  { { .i=-1234567890123456 }, BE_INT},
  { { .p=(void*)0x4000000000000000 }, BE_REAL},
  { { .p=(void*)0x3FA111111111110E }, BE_REAL},
  { { .s=be_local_const_str(f_str_3) }, BE_STRING},
};

static const uint32_t f_code[7] = {
  0x6008000F,  //  0000  GETGBL	R2	G15
  0x580C0000,  //  0001  LDCONST	R3	K0
  0x7C080200,  //  0002  CALL	R2	1
  0x6008000F,  //  0003  GETGBL	R2	G15
  0xC0E0302,  //  0004  DIV	R3	R257	R258
  0x7C080200,  //  0005  CALL	R2	1
  0x80060600,  //  0006  RET	1	R259
};

static const bproto f_proto = {
  NULL,     // bgcobject *next
  8,       // type
  GC_CONST,        // marked
  4,       // nstack
  0,       // nupvals
  2,       // argc
  0,       // varg
  NULL,     // bgcobject *gray
  NULL,     // bupvaldesc *upvals
  (bvalue*) &f_ktab,     // ktab
  NULL,     // bproto **ptab
  (binstruction*) &f_code,     // code
  be_local_const_str(f_str_name),       // name
  7,       // codesize
  4,       // nconst
  0,       // nproto
  be_local_const_str(f_str_source),     // source
#if BE_DEBUG_RUNTIME_INFO /* debug information */
  NULL,     // lineinfo
  0,        // nlineinfo
#endif
#if BE_DEBUG_VAR_INFO
  NULL,     // varinfo
  0,        // nvarinfo
#endif
};

const bclosure f_closure = {
  NULL,     // bgcobject *next
  36,       // type
  GC_CONST,        // marked
  0,       // nupvals
  NULL,     // bgcobject *gray
  (bproto*) &f_proto,     // proto
  { NULL }     // upvals
};

/*******************************************************************/
```

Skiars, 2021-03-24:
The `bproto` can be a constant. If you want to support upvalues, sub-proto, etc., you may need to instantiate the closure at runtime. I think this RAM usage is not large (proto is constant).

If a function does not have an upvalue, then we can instantiate it as a closure in the coc phase, just like you do now:

``` c
const bclosure f_closure = {
  NULL,     // bgcobject *next
  36,       // type
  GC_CONST,        // marked
  0,       // nupvals
  NULL,     // bgcobject *gray
  (bproto*) &f_proto,     // proto
  { NULL }     // upvals
};
```

If it is a sub-proto, but there are no upvalues, we can use LDCONST to replace CLOSURE instruction (you have to push the constant closure into the k-tab of the proto). But there is no good way for the global object.

s-hadinger, 2021-03-25:
Thanks for your insights. Right now I don't need the subproto so I can postpone it. I need to fix some detail in the flash strings hashcodes.

Overall I have moved all the berry code in Tasmota to solidified and it works great. I went from 10K to 4k of RAM.

Skiars, 2021-03-25:
Well, these are just some of my ideas, we can implement them later.

s-hadinger, 2021-03-26:
Sure, and thanks for your advice. The current form is a bit ugly, Maybe some macros can help make it better looking.

s-hadinger, 2021-04-07:
I finally added support for subprotos and for upvals. I also created some macros to make the result less ugly. It's in production in Tasmota currently.

Here is the branch: https://github.com/s-hadinger/berry/tree/solidify_1

It will need some more testing and cleaning.

Example:
```
class test
  var dummy
  def f(m)
    return /-> print(m)
  end 
end
t = test()

import solidify
solidify.dump(t.f)
```

Produces:
```
/********************************************************************
** Solidified function: f
********************************************************************/

/********** Solidified proto: f_0 */
static const bupvaldesc f_0_upvals[1] = {
  be_local_const_upval(1, 1),
};

be_define_local_const_str(f_0_str_name, "<lambda>", 607256038, 0, 8, 0);
be_define_local_const_str(f_0_str_source, "stdin", -1529146723, 0, 5, 0);

static const uint32_t f_0_code[4] = {
  0x6000000F,  //  0000  GETGBL R0 G15
  0x68040000,  //  0001  GETUPV R1 U0
  0x7C000200,  //  0002  CALL R0 1
  0x80040000,  //  0003  RET 1 R0
};

be_define_local_proto(f_0, 2, 0, 0, 0, 1);

/********** Solidified proto: f */
static const bproto *f_subproto[1] = {
  &f_0_proto,
};

be_define_local_const_str(f_str_name, "f", -485742695, 0, 1, 0);
be_define_local_const_str(f_str_source, "stdin", -1529146723, 0, 5, 0);

static const uint32_t f_code[3] = {
  0x84080000,  //  0000  CLOSURE R2 P0
  0xA0000000,  //  0001  CLOSE 0
  0x80040400,  //  0002  RET 1 R2
};

be_define_local_proto(f, 3, 2, 0, 1, 0);
be_define_local_closure(f);

/*******************************************************************/
```

s-hadinger, 2021-04-29:
Closing since it's now provided my `solidify` module.

closed 2021-04-29 by s-hadinger