berry-lang/berry #84: Do you have a website ?

diazvictor, 2021-03-11:
I love it ❤️, if you don't have a website I'm willing to build it.

Skiars, 2021-03-12:
Very welcome!

I have made a Berry interpreter that can be run REPL on a [web page](https://skiars.github.io/berry_web/). This project may be useful: https://github.com/Skiars/berry_web

diazvictor, 2021-03-13:
Excellent, I will send you a demo soon. 👌

diazvictor, 2021-03-19:
> Very welcome!
> 
> I have made a Berry interpreter that can be run REPL on a [web page](https://skiars.github.io/berry_web/). This project may be useful: https://github.com/Skiars/berry_web

How are you doing, here is the demo I promised, it still lacks content and improve the examples, tell me if you like it to know if it is worth to continue working on it, if so let me know if you want to add/remove something. Thanks

https://github.com/diazvictor/berry_web/

Skiars, 2021-03-19:
In fact, I use [Emscripten](https://developer.mozilla.org/en-US/docs/Mozilla/Projects/Emscripten) to compile Berry into js for execution on the web side. Of course, we can also use wasm.

I think [Haskell](https://www.haskell.org/)'s pages are doing well. Although Berry doesn't have that much content, we can still list the main features in a suitable layout. Finally, an online REPL may be a nice feature.

All in all, I think the page can be very simple, but at the same time it allows users to try something.

MasterVitronic, 2021-03-19:
Hi, I think @diazvictor  spoke but forgot to show the demo, here is the demo for your consideration. https://diazvictor.github.io/berry_web/

Skiars, 2021-03-20:
It looks pretty cool. I think this page is enough.

Skiars, 2021-03-20:
From my perspective, I hope to improve the "try online" function. But this may not be easy.

diazvictor, 2021-03-22:
> It looks pretty cool. I think this page is enough.

Hi, I have finished the website, please [take a look at it](https://diazvictor.github.io/berry_web/), especially the "tour" section regarding the presentation.

I want to discuss the possibility of creating an organization like "berry-lang", to have a good domain on github. for example berry-lang.github.io. This way I can index the page to rank well in google and other search engines.

I would also like to maintain the page if you agree.

s-hadinger, 2021-03-22:
This looks great!

Is it possible to add a section mentioning other projects using Berry, and add Tasmota?

It's now officially part of our project: https://tasmota.github.io/docs/Berry-Scripting/

Main page for Tasmota: https://github.com/arendst/Tasmota

Skiars, 2021-03-22:
I saw everyone’s work and it was really exciting!

If we are sure to do more work, documentation, examples and tutorials may need to be improved. I think I did not do well in this regard.

s-hadinger, 2021-03-22:
The documentation needs some updates. I'm just curious, what are you using to convert from Latex to PDF?

MasterVitronic, 2021-03-22:
We have also created a group on Telegram,
please feel welcome to participate in a more dynamic way through this medium.
https://t.me/berry_lang

diazvictor, 2021-03-23:
> We have also created a group on Telegram,
> please feel welcome to participate in a more dynamic way through this medium.
> https://t.me/berry_lang

I also recommend activating the "Discussions" section, so as not to have to create a new "Issue" to discuss "X" topic

Skiars, 2021-03-24:
> The documentation needs some updates. I'm just curious, what are you using to convert from Latex to PDF?

I use [TeXLive](http://tug.org/texlive/).

hiperiondev, 2022-10-24:
This issue is still relevant or have to be closed?

closed 2024-04-15 by diazvictor