berry-lang/berry #85: Bug in list concatenation

s-hadinger, 2021-03-12:
When adding a list to a list, the first operand is modified.

As you see below, when doing `a+b` the result is stored in `a`. I suppose that a new list should be allocated rather than modifying the first one.

```
> a=[1]
> b=[2]
> a+b
[1, 2]
> a
[1, 2]
> b
[2]
```

The same problems happens with `..`

```
> c=[3]
> c..4
[3, 4]
> c
[3, 4]
```

Skiars, 2021-03-13:
The `+` operator really shouldn't modify the operands, so this is a bug. But `..` is not. I expect it to be similar to `<<` in C++, and its left operand is modifiable. To perform quick string splicing and other operations:

```
> ([]..1..2..{'a':true}..6).concat() 
12{'a': true}6
```

s-hadinger, 2021-03-13:
Thanks, this was not clear to me. Makes sense.

Though it's not consistent with the effect of `..` on strings:

```
> s='a'
> s..'b'
'ab'
> s
'a'
```

The string s should be modified, right?

closed 2021-03-13 by s-hadinger