berry-lang/berry #91: command line arguments

MasterVitronic, 2021-03-21:
Greetings @Skiars , I wonder if you have any
interest in adding similar support to :

`berry script.be arg1 arg2`

I'm trying to use it as a general purpose language, and I'm missing these features

Skiars, 2021-03-22:
You can use the global variable `_argv`:

```
print(_argv)
```

MasterVitronic, 2021-03-22:
how interesting, I thought I had checked the documentation well, thank you very much, now I feel silly because it seems obvious, if you don't mind, I will make a pull request on the following example:

```ruby
#!/usr/bin/env berry
#- file cli_args.be -#

import string

if _argv.size() <= 1 
	print(string.format('Usage: %s argv1 argv2', _argv[0]))
	return false
end

for i : 0 .. _argv.size() - 1
    print(i, '\t=>', _argv[i])
end
```

Skiars, 2021-03-22:
Ok!😄

closed 2021-03-24 by Skiars