berry-lang/berry #93: Introspection: calling an object's method by name

s-hadinger, 2021-03-22:
I would like to call an object method by its name passed as a string. I don't believe it is possible right now in Berry code. Did I miss it?

It would serve as a dispatcher to call a method whose name is the same as the name of the event passed to it.

If not, I'm happy to add an `introspect` module or any other form you prefer to add it. It would basically call `be_getmethod()` with a string argument. Which you be your preferred way?

Skiars, 2021-03-23:
This feature is not currently supported. In addition to calling library functions, can it be achieved through syntax like this?

```
obj.(expr) # the value of expr is a string type.
```

s-hadinger, 2021-03-23:
That would be a nice addition.

As it impacts the lexer and compiler, I can't assess the impact to the code, whereas an additional fonction is easy to do.

Skiars, 2021-03-23:
To support this grammar is very simple, I can do it on weekends.

s-hadinger, 2021-03-23:
Would it work for globals as well:

```
def log(s) print(s) end
l="log"
(l)('hello')
```

This would solve the issue of accessing globals from solidified code.

Skiars, 2021-03-24:
This is a question to consider, I didn't expect it before.

closed 2021-09-13 by s-hadinger