berry-lang/berry #100: Documentation in Markdown?

s-hadinger, 2021-05-27:
Hi @Skiars,
The Berry documentation is very high quality, but is starting to be outdated with the latest additions. Latex is too complex for many users (including me) to edit and update.
Would you consider moving the documentation to Markdown in Github to make Pull Requests and contribution easier?

Skiars, 2021-05-28:
I support this idea. But the migration of documents is a lot of work, and I don't have much time now.

diazvictor, 2021-05-28:
> I support this idea. But the migration of documents is a lot of work, and I don't have much time now.

I am willing to collaborate with the markdown migration, if desired

Skiars, 2021-05-28:
> > I support this idea. But the migration of documents is a lot of work, and I don't have much time now.
> 
> I am willing to collaborate with the markdown migration, if desired

Great!

s-hadinger, 2021-06-29:
@Skiars Would you be kind enough to open the wiki feature to all users so we can edit the documentation here?

Skiars, 2021-06-29:
The wiki is already open.

s-hadinger, 2021-06-29:
Oh yes sorry

s-hadinger, 2021-08-11:
I did a first pass on the Reference Manual with pandoc and some manual editing. There are still some glitches but it’s mostly usable. 

https://github.com/Skiars/berry/wiki/Reference

Skiars, 2021-08-13:
I took a look at the documentation, and I think it is really usable.

closed 2021-09-13 by s-hadinger