berry-lang/berry #111: Proposed extension: class constants

s-hadinger, 2021-07-13:
I've been using Berry for a while now and I'm missing a feature that would be a nice addition: class constants.

Currently the only way to add constants is to put them in a module. I would love to have an equivalent to declare class-wide constants.

Let's say we want to declare two constants (equivalent of C enums): ON=1 OFF=0

Currently the only way is:

```python
class A
  var ON, OFF
  def init()
    self.ON = 1
    self.OFF = 0
  end
  def so_some_stuff()
    print(self.ON)
  end
end
```

Unfortunately this is cumbersome as it needs explicit code in `init()` and consumes memory for each instance.

I'm proposing the additional syntax:

```python
class A
  const ON = 1
  const OFF = 0
  def so_some_stuff()
    print(self.ON)
  end
end
```

This would imply:
- adding `const` as a keyword
- fix parser to accept the new syntax
- add const values to `bclass.members` (hoping this is enough)

@Skiars would you be ok with this addition?

s-hadinger, 2021-07-14:
I've been trying and realized it was almost there. 

The internal structure does not need any change, `bclass->members` is already able to contain values. The only problem is that `MT_VARIABLE` is mapped to `BE_INT` which prevents storing int values (or they are confused with the index for the instance variable.

I propose to create a new explicit `BE_VAR` type that will be used for `MT_VARIABLE` so `BE_INT` can be used in classes for int constants.

Also I realized that the syntax won't work. Setting the class variables cannot be done by the parser and needs code to be executed. So the alternative I propose is to change `const` to `static`:

```python
class A
  var x
  static ON, OFF
  def so_some_stuff()
    print(self.ON)
  end
end

A.on = 1
A.off = 0
```

And for pre-compiled classes it looks like:
```
/* @const_object_info_begin
class be_class_A (scope: global, name: A) {
    x, var
    ON, int(1)
    OFF, int(0)
[...]
}
@const_object_info_end */
```

Skiars, 2021-07-14:
I want to add the syntax for initializing static members in the class declaration, like:

``` python
class A
  static ON = 1, OFF = 0
end
```

I even want to be able to initialize class member variables at the time of declaration:

``` python
class A
  var a = "hello"
end
```

I think this is feasible.

In other words, static members and methods are very similar, we have to consider this:

``` python
class A
  static a = def () print('xxx') end
end
```

This will be mistaken for the method.

s-hadinger, 2021-07-14:
Thanks, I will have soon a working version for `static` class variables.

I'm not sure how you want to implement initializers. Either they are limited to simple types, or you allow code to be called.

The following example requires code to run, and `now()` could be a time function that returns different values with different instances.

```
class A
  var a = 1+1
  var b = now()
end
```

In such case, I can only see this as syntactic sugar for the equivalent:

```
class A
  var a = 1+1
  var b = now()
  def _preinit()
    self.a = 1+1
    self.b = now()
  end
  def init()
    self._preinit()
  end
end
```

And it may be a little more subtle to handle inheritance. Is this what you want?

Skiars, 2021-07-14:
You can implement the static member first, but you need to consider that the static member type may be a function, etc.

I think the inline initialization of members needs more consideration. This brings many challenges:
- I tend to generate the code directly for the initialization, and then splice it before the code of the constructor;
- Need to pay attention to whether the instruction address in the code will be destroyed (luckily, we always use the jump relative to PC);
- In the inline initialization statement, the self parameter is not available;
- The base class constructor should be explicitly called in the constructor, it is always called after the inline initialization code (in C++, first construct the base class, then construct the members of the class, and finally execute the constructor).

Inline initialization of static members is different. These codes should be generated into the function that declares the class.

Skiars, 2021-07-14:
Two types of values ​​are stored in `bclass->members`: index of member variables, and function (closure or native closure). Therefore, `MT_VARIABLE` is mapped to `BE_INT`.

I think it is not feasible to use `class->members` to store static members directly. Obviously, it will not be able to distinguish static member variables of methods and stored functions.

s-hadinger, 2021-07-14:
I have a working version with static members using the current `members` map. I will submit a PR tomorrow.

Actually not being able to distinguish a function from a method is not necessarily a problem.

From what I understand, Berry uses the calling convention to see if it's a method or not:

```python
class A
  def f(x) return type(self) end    # return the type of `self`
end
a=A()

# call f() as a method, first argument is the instance
> a.f(0)
'instance'

# we copy the method to a function and call it as a function
> f2 = a.f
> f2(0)    # here no `self` is implicitly inserted as first argument
'int'
```

So the same code might be called as a method or as a function depending on the caller code. With this, it would be normal that a static function may be used as a method:
```
> class A static f end
> A.f = def(self, x) return type(self) end

> a = A()
> a.f(0)
'instance'
```

The above is actually the behavior I would expect.

Skiars, 2021-07-15:
Well, it makes sense. I agree.

s-hadinger, 2021-07-15:
Done, the PR is out.

s-hadinger, 2021-07-15:
I have a clear vue about how to add inline initialization of static variables. But I don't see how to do instance variable inline initialization (yet).

Skiars, 2021-07-19:
This can be discussed in a new issue.

closed 2021-07-29 by s-hadinger