berry-lang/berry #119: Proposal for vararg functions

s-hadinger, 2021-08-11:
Hi @Skiars 

I saw that you had vararg functions in the roadmap and it does not seem too difficult to implement. I saw that you already have `varg` field in `bproto`

## Proposed syntax
Vararg is already supported by native functions so we only need a specific implementation for Berry closures. It shouldn’t make any difference from the caller perspective.

Normal parameters are passed unchanged, vararg are grouped in a list passed as additional parameter.

I propose to use `...` syntax like this:

Option 1 

```
def f(a, b, ...)
  print(a, b)
  for i:_va
    print(i)
  end
end
```

This option is similar to C but requires to define a convention to get additional parameters. The proposed additional implicit local variable would be `_va`

Option 2

```
def g(a, b, rest...)
  print(a, b)
  for i:rest
    print(i)
  end
end
```

This second option is different from C, but allows to explicitly name the arg containing the list. 

I tend to prefer option 2 with the risk of being too different from C convention. 

Any preference?

s-hadinger, 2021-08-11:
## Implementation 

I identified the following changes needed:

- add a new token `...` in the lexer 
- change the parser of function args to detect this token and mark the `bproto` accordingly 
- btw `varg` in `bproto` only needs to be a flag since the vararg parameter is always the last one
- patch VM CALL to build the list of vararg at runtime

Skiars, 2021-08-13:
I tend to be similar to the python syntax:
```
def func(a, b, *c)
     ...
```
So no new symbols are needed.

Another issue I am more worried about is that the support of variable arguments will reduce the performance of VM CALL.

s-hadinger, 2021-08-13:
That’s a neat syntax.

It would add a simple test on proto->varg so there would be no performance hit for regular functions. Only functions defined as vararg would indeed incur the creation and population of the `list` object. I don’t think it’s a major issue and I don’t see any other way to do it since registers are statically allocated in VM compiled code. 

Let’s try

closed 2021-09-13 by s-hadinger