berry-lang/berry #122: Example of calling a berry function from the host ( C api )

JettMonstersGoBoom, 2021-08-17:
Is it possible to call a berry function from C and if so , could you share an example ?

Skiars, 2021-08-18:
Use `be_call`. For the documentation of C interface, please refer to here: https://github.com/Skiars/berry/wiki/Chapter-8

I suggest you read the code in the [default](https://github.com/Skiars/berry/tree/master/default) folder and [berry.h](https://github.com/Skiars/berry/blob/master/src/berry.h).

The following resources are also available for reference:
- https://github.com/Skiars/berry_regex
- https://github.com/Skiars/berry/blob/master/src/be_mathlib.c
- https://github.com/Skiars/berry/blob/master/src/be_timelib.c

JettMonstersGoBoom, 2021-08-19:
Thanks . incase anyone is interested. 
```
// load the script
int res = be_loadmode(vm,"main.be",false);
handle_result(vm, res); // from berry.c 
res=be_pcall(vm,0); // this was the line I was missing 
```
then to call a function
```
   res=be_loadstring(vm, "functionFromMain(1)"); // Compile test code
   handle_result(vm, res);
   res=be_pcall(vm,0);
   handle_result(vm, res);
```

closed 2021-08-19 by JettMonstersGoBoom