berry-lang/berry #144: Unstable builtin internal number

s-hadinger, 2021-09-13:
Introducing `call()` created a disruption in my pre-solidifed code. I discovered that the builtins internal numbers are stable in non-precompiled berry but are not stable with pre-compiled code.

Non-precompiled (print has index 1 which corresponds to its index in `be_load_baselib()`:
```
> def f() return print end
> import debug
> debug.codedump(f)
source 'stdin', function 'f':
; line 1
  0000  GETGBL	R0	G1
  0001  RET	1	R0
```

Pre-compiled:
```
> def f() return print end
> import debug
> debug.codedump(f)
source 'stdin', function 'f':
; line 1
  0000  GETGBL	R0	G15
  0001  RET	1	R0
```

In pre-compiled berry, it looks like the index is hashed? by coc compiler. Anyways their indexes are not the same as non-compiled code, and introducing `call()` as additional function reshuffles all existing indixes.

Unfortunately this breaks solidified code and compiled bytecode.

I don't know enough coc compiler, is there any way to force keeping the order of definitions in `vartab`?

Skiars, 2021-09-17:
This should be because `coc` did not consider this. You can look at the `generate\be_fixed_m_builtin.h` file. Obviously, the value of `m_builtin_map` corresponds to the number of the function pointer in `__vlist_array`. 

I am sure this order can be adjusted to meet your needs.

s-hadinger, 2021-09-17:
Thanks. I solved in in Coc. I will send a PR

closed 2021-09-21 by s-hadinger