berry-lang/berry #165: `print(5)` only work on one platform

paulrouget, 2021-11-09:
```
print(5)
```

I compile the script to bytecode on Mac (M1 chip).
I run on Mac, it prints "5".
I run on a Cortex M4, it prints "ld".

`print("ok")` works on both platform.

The bytecode is supposed to be multiplatform?

paulrouget, 2021-11-09:
I guess it's an endianness issue.

s-hadinger, 2021-11-09:
I doubt it. Intel and ARM macs are both little endian. The bytecode is not intended to be multiplatform because it depends on compile options like size of int and size of float. However if you compile with the same options it should work. 

Can you share the compiled bytecode on both platforms?

paulrouget, 2021-11-09:
Mac and Cortex are both ARM. Not sure if that matters or not…

I don't have a compiler for the Cortex M4.

I generate the bytecode from my Mac:

Bytecode (base64):

```sh
echo "vs3+AgMAAAAXAAAAAQAAAAUAX2FyZ3YEAG1haW4HAHRlc3QuYmUAAgAABAAAAAEAAGAEAAZUAAIAfAAAAIAAAAAAAAAAAAA=" | base64 -d > /tmp/test.o
```

berry_conf.h:

```c
/********************************************************************
** Copyright (c) 2018-2020 Guan Wenliang
** This file is part of the Berry default interpreter.
** skiars@qq.com, https://github.com/Skiars/berry
** See Copyright Notice in the LICENSE file or at
** https://github.com/Skiars/berry/blob/master/LICENSE
********************************************************************/
#ifndef BERRY_CONF_H
#define BERRY_CONF_H

#include <assert.h>

/* Macro: BE_DEBUG
 * Berry interpreter debug switch.
 * Default: 0
 **/
#ifndef BE_DEBUG
#define BE_DEBUG                        0
#endif

/* Macro: BE_LONGLONG_INT
 * Select integer length.
 * If the value is 0, use an integer of type int, use a long
 * integer type when the value is 1, and use a long long integer
 * type when the value is 2.
 * Default: 2
 */
#define BE_INTGER_TYPE                  2

/* Macro: BE_USE_SINGLE_FLOAT
 * Select floating point precision.
 * Use double-precision floating-point numbers when the value
 * is 0 (default), otherwise use single-precision floating-point
 * numbers.
 * Default: 0
 **/
#define BE_USE_SINGLE_FLOAT             0

/* Macro: BE_USE_PRECOMPILED_OBJECT
 * Use precompiled objects to avoid creating these objects at
 * runtime. Enable this macro can greatly optimize RAM usage.
 * Default: 1
 **/
#define BE_USE_PRECOMPILED_OBJECT       1

/* Macro: BE_DEBUG_RUNTIME_INFO
 * Set runtime error debugging information.
 * 0: unable to output source file and line number at runtime.
 * 1: output source file and line number information at runtime.
 * 2: the information use uint16_t type (save space).
 * Default: 1
 **/
#define BE_DEBUG_RUNTIME_INFO           1

/* Macro: BE_DEBUG_VAR_INFO
[... 116 lines cut ...]
#define BE_USE_GC_MODULE                1
#define BE_USE_SOLIDIFY_MODULE          1
#define BE_USE_INTROSPECT_MODULE        1
#define BE_USE_STRICT_MODULE            1

/* Macro: BE_EXPLICIT_XXX
 * If these macros are defined, the corresponding function will
 * use the version defined by these macros. These macro definitions
 * are not required.
 * The default is to use the functions in the standard library.
 **/
#define BE_EXPLICIT_ABORT               abort
#define BE_EXPLICIT_EXIT                exit
#define BE_EXPLICIT_MALLOC              malloc
#define BE_EXPLICIT_FREE                free
#define BE_EXPLICIT_REALLOC             realloc

/* Macro: be_assert
 * Berry debug assertion. Only enabled when BE_DEBUG is active.
 * Default: use the assert() function of the standard library.
 **/
#define be_assert(expr)                 assert(expr)

#endif
```

s-hadinger, 2021-11-09:
Sorry I thought it was both macs. Modern processors are often capable of doing both little and big endian, it depends on how they are configured at boot. But indeed bytecode will be different if endianness is different.

For Cortex M4 I would suggest to use 32 bits ints and 32 bits floats, it will probably make smaller code.

How do you test on Cortex M4 if you don't have a compiler?

paulrouget, 2021-11-09:
I generate the bytecode on mac.

paulrouget, 2021-11-09:
And the only use the vm on the Cortex.

paulrouget, 2021-11-10:
By compiling I meant compiling my berry code into bytecode.

The goal is to avoid extra work on the embedded device, and also save space.
Compiling require extra CPU work, and using source code instead of bytecode use extra Flash space. It's work that can be done ahead of time on a host machine.

It's something I was doing with Lua. There is a cross-compiler that allows compiling for the embedded CPU from a host machine.

Would that be something doable?

paulrouget, 2021-11-10:
Hmm, it looks like something else. I'm now not using bytecode on the device, but the source directly, and I get the same result.

I'm investigating…

paulrouget, 2021-11-10:
```
print("foo")
print("5")
print(5)
print("8")
print(8)
print("bar")
```

This prints:

```
foo
5
ld
8
ld
bar
```

I use `be_loadmode(vm, "test.be", false)`

paulrouget, 2021-11-10:
Using `BE_INTGER_TYPE=1` instead of `2` solves the issue.

But I'd like to use 64 bits integers.

paulrouget, 2021-11-10:
Results:

- `BE_INTGER_TYPE=2`: compiled on Mac or Cortex, executed on Cortex, I'm getting "ld" when calling `print(5)`.
- `BE_INTGER_TYPE=2`: compiled on Mac, executed on Mac, it works as expected.
- `BE_INTGER_TYPE=1`: compiled on Cortex, executed on Cortex, it works as expected.
- `BE_INTGER_TYPE=1`: compiled on Mac, executed on Cortex, I get an "invalid bytecode" error.
- `BE_INTGER_TYPE=0`: compiled on Mac, executed on Cortex, it works as expected.

s-hadinger, 2021-11-10:
I think that your cortex M4 libs does not support 64 bits. Which is common on embedded devices to save space. 

As said before I strongly advise to use 32 bits integers. If you use the same Berry compile options then you can do cross bytecode compile

paulrouget, 2021-11-10:
Understood.

`BE_INTGER_TYPE=0` works well.

I'm using Berry on [this chip](https://www.nordicsemi.com/products/nrf52832) from Rust code, and it's working really well. Lot nicer than using lua or other solutions.

Thank you.

closed 2021-11-10 by paulrouget