berry-lang/berry #182: How to load multiple files with one VM

paulrouget, 2021-12-13:
I have 2 scripts:

foo.be:

```ruby
def test()
  print("foo")
end
```

bar.be:

```ruby
def test()
  print("bar")
end
```

On my laptop, I compile these scripts into bytecode:

```shell
berry -c foo.be -i foo.o
berry -c bar.be -i bar.o
```

On my MCU, I create **one** VM and load the byte code (not using the compiler on the MCU):

```c
be_loadmode(vm, file, false)
```

I want to be able to call foo::test and bar::test from C code.

What's the proper approach?

For now I just create 2 VMs, but I imagine that's not the best approach.

s-hadinger, 2021-12-13:
You can't have two functions with the same global name and you should have 2 different names for both functions. There is no concept of namespace per se.

The good way to do this would be to create a module:

``` ruby
foo = module("foo")
foo.test = def()
  print("foo")
end

return foo
```

``` ruby
bar = module("bar")
bar.test = def()
  print("bar")
end

return bar
```

Then load with:
``` ruby
import foo
import bar

foo.test()
bar.test()
```

paulrouget, 2021-12-14:
Understood. Thank you.

closed 2021-12-14 by paulrouget