berry-lang/berry #201: This code segfaults every ~10 runs

paulrouget, 2021-12-14:
```ruby
def foo(x)
  for i: 0 .. 10
    if i == 42
    end
  end
end
```
Running this code will segfault every ~10 runs. No need to excute `foo`.

Tested with 3d39a9e.

s-hadinger, 2021-12-14:
I'm seeing also some segfault on Mac randomly and I never found where it comes from. I tried using a debugger but didn't find the source of the crash.

It is however rock-solid on ESP32. So I suspect something related to unix port being the cause of the crash.

What is your platform?

paulrouget, 2021-12-14:
Happening at compile time. Compiling on Mac. Running on nRF52.

paulrouget, 2021-12-14:
It's surprising that it's not deterministic.

s-hadinger, 2021-12-14:
I agree. I tried enabling GC debug, but it's not related. I tried increasing the Berry stack at start to avoid stack resize (it caused crashes in the past) and it didn't solve it. I tried removing `readline` library that caused other issues, it didn't solve it.

Do you have any hint where the crash occurs?

paulrouget, 2021-12-14:
> Do you have any hint where the crash occurs?

I think it's [here](https://github.com/berry-lang/berry/blob/ca5f928efad9241f4b563afdf2182a62956507db/src/be_parser.c#L177). But no idea why.

s-hadinger, 2021-12-14:
Excellent finding. I don't use `BE_DEBUG_VAR_INFO` on ESP32, this is why I don't have any crash.

I disabled `#define BE_DEBUG_VAR_INFO 0` on mac and segfaults seem to be gone.

We can now inspect why there is this crash.

paulrouget, 2021-12-14:
Indeed, I was using `BE_DEBUG_VAR_INFO` on Mac (but not on my MCU).

Skiars, 2021-12-14:
Regarding this kind of problem, I suggest using Address Sanitize, adding `-fno-omit-frame-pointer -fsanitize=address` to `CFLAGS` and `LFLAGS` (support GCC or Clang):

[image]

And then I get:
```
/home/gwl/Documents/code/berry/berry tests/issue-201.be
=================================================================
==240564==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x606000000428 at pc 0x56401955c742 bp 0x7ffe1bef0b50 sp 0x7ffe1bef0b40
READ of size 4 at 0x606000000428 thread T0
    #0 0x56401955c741 in end_varinfo src/be_parser.c:177
    #1 0x56401955ccac in end_block_ex src/be_parser.c:216
    #2 0x56401955ce45 in end_block src/be_parser.c:224
    #3 0x564019568bc1 in block src/be_parser.c:1705
    #4 0x5640195641ea in condition_block src/be_parser.c:1143
    #5 0x564019564351 in if_stmt src/be_parser.c:1156
    #6 0x5640195688a8 in statement src/be_parser.c:1674
    #7 0x564019568aa1 in stmtlist src/be_parser.c:1696
    #8 0x564019564cf1 in for_iter src/be_parser.c:1229
    #9 0x564019565401 in for_stmt src/be_parser.c:1274
    #10 0x5640195688ca in statement src/be_parser.c:1676
    #11 0x564019568aa1 in stmtlist src/be_parser.c:1696
    #12 0x5640195600d5 in funcbody src/be_parser.c:611
    #13 0x564019565b57 in def_stmt src/be_parser.c:1362
    #14 0x56401956890b in statement src/be_parser.c:1680
    #15 0x564019568aa1 in stmtlist src/be_parser.c:1696
    #16 0x564019568f19 in mainfunc src/be_parser.c:1717
    #17 0x5640195692b5 in be_parser_source src/be_parser.c:1735
    #18 0x564019540c7d in m_parser src/be_exec.c:158
    #19 0x5640195403ff in be_execprotected src/be_exec.c:116
    #20 0x564019540ef1 in be_protectedparser src/be_exec.c:175
    #21 0x564019541313 in fileparser src/be_exec.c:218
    #22 0x56401954166b in be_loadmode src/be_exec.c:262
    #23 0x564019588261 in doscript default/berry.c:205
    #24 0x56401958832d in load_script default/berry.c:223
    #25 0x564019588b87 in analysis_args default/berry.c:337
    #26 0x564019588cb8 in main default/berry.c:365
    #27 0x7f23f1b92b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)
    #28 0x56401951e66d in _start (/home/gwl/Documents/code/berry/berry+0x2666d)

0x606000000428 is located 8 bytes to the right of 64-byte region [0x6060000003e0,0x606000000420)
allocated by thread T0 here:
    #0 0x7f23f1f8f652 in __interceptor_realloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:164
    #1 0x564019557e39 in _realloc src/be_mem.c:50
    #2 0x564019557f04 in be_realloc src/be_mem.c:67
    #3 0x564019576db0 in be_vector_push src/be_vector.c:42
    #4 0x564019576f45 in be_vector_push_c src/be_vector.c:58
    #5 0x56401955c3b9 in begin_varinfo src/be_parser.c:159
    #6 0x56401955e99f in new_localvar src/be_parser.c:413
    #7 0x564019564c68 in for_iter src/be_parser.c:1226
    #8 0x564019565401 in for_stmt src/be_parser.c:1274
    #9 0x5640195688ca in statement src/be_parser.c:1676
    #10 0x564019568aa1 in stmtlist src/be_parser.c:1696
    #11 0x5640195600d5 in funcbody src/be_parser.c:611
    #12 0x564019565b57 in def_stmt src/be_parser.c:1362
    #13 0x56401956890b in statement src/be_parser.c:1680
    #14 0x564019568aa1 in stmtlist src/be_parser.c:1696
    #15 0x564019568f19 in mainfunc src/be_parser.c:1717
    #16 0x5640195692b5 in be_parser_source src/be_parser.c:1735
    #17 0x564019540c7d in m_parser src/be_exec.c:158
    #18 0x5640195403ff in be_execprotected src/be_exec.c:116
    #19 0x564019540ef1 in be_protectedparser src/be_exec.c:175
    #20 0x564019541313 in fileparser src/be_exec.c:218
    #21 0x56401954166b in be_loadmode src/be_exec.c:262
    #22 0x564019588261 in doscript default/berry.c:205
    #23 0x56401958832d in load_script default/berry.c:223
    #24 0x564019588b87 in analysis_args default/berry.c:337
    #25 0x564019588cb8 in main default/berry.c:365
    #26 0x7f23f1b92b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)

SUMMARY: AddressSanitizer: heap-buffer-overflow src/be_parser.c:177 in end_varinfo
Shadow bytes around the buggy address:
  0x0c0c7fff8030: fa fa fa fa 00 00 00 00 00 00 00 00 fa fa fa fa
  0x0c0c7fff8040: fd fd fd fd fd fd fd fd fa fa fa fa 00 00 00 00
  0x0c0c7fff8050: 00 00 00 00 fa fa fa fa 00 00 00 00 00 00 00 00
  0x0c0c7fff8060: fa fa fa fa 00 00 00 00 00 00 00 00 fa fa fa fa
  0x0c0c7fff8070: 00 00 00 00 00 00 00 00 fa fa fa fa 00 00 00 00
=>0x0c0c7fff8080: 00 00 00 00 fa[fa]fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff8090: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff80a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff80b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff80c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff80d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==240564==ABORTING
```

s-hadinger, 2021-12-14:
Wow, I didn't know this option. It confirms the location of the bug.

s-hadinger, 2021-12-14:
I think it's a simple patch, there is potentially a pointer going wild if `it>end` but the end condition not matched.

My fix proposal is:

```
    for (; (it->beginpc < beginpc) && (it <= end); ++it);
```

s-hadinger, 2021-12-14:
This bug seems fixed, but there are some others in `end_block`

s-hadinger, 2021-12-14:
I have a new simple test case that creates a stack corruption:

```
class A
    def f(n)
        if (n == 'f1') return / n -> n end
    end
end
```

With linux, I get the following trace:

```
==22505==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6030000010e8 at pc 0x000000448d35 bp 0x7fff6b3e74a0 sp 0x7fff6b3e7
490
READ of size 4 at 0x6030000010e8 thread T0
    #0 0x448d34 in end_varinfo src/be_parser.c:177
    #1 0x448dd9 in end_block_ex src/be_parser.c:216
```

s-hadinger, 2021-12-14:
Stupid me, the test must be reversed to avoid dereferencing an invalid pointer:

```
    for (; (it <= end) && (it->beginpc < beginpc); ++it);
```

s-hadinger, 2021-12-14:
Solved. There is another issue in bytes but that's another problem (double free on a pointer)

s-hadinger, 2021-12-14:
Double free is solved as well. Now all tests pass with `-fno-omit-frame-pointer -fsanitize=address`

paulrouget, 2021-12-15:
Thanks a lot!

closed 2021-12-15 by paulrouget