berry-lang/berry #207: Can't compile if `BE_USE_PERF_COUNTERS` set to `0`

paulrouget, 2021-12-15:
```
src/be_gc.c:564:82: error: no member named 'counter_gc_kept' in 'struct bvm'
    if (vm->obshook != NULL) (*vm->obshook)(vm, BE_OBS_GC_END, vm->gc.usage, vm->counter_gc_kept, vm->counter_gc_freed);
                                                                             ~~  ^
src/be_gc.c:564:103: error: no member named 'counter_gc_freed' in 'struct bvm'
    if (vm->obshook != NULL) (*vm->obshook)(vm, BE_OBS_GC_END, vm->gc.usage, vm->counter_gc_kept, vm->counter_gc_freed);
                                                                                                  ~~  ^
2 errors generated.
```

paulrouget, 2021-12-15:
Same goes for `BE_DEBUG_VAR_INFO`.

paulrouget, 2021-12-15:
I think the recent introduction to pycoc breaks `berry_conf.h`. Apparently it's improperly scanning the headers. Not taking into account any `#ifndef` blocks and such.

Skiars, 2021-12-15:
Do we need to support a full C preprocessor?

paulrouget, 2021-12-15:
This issue (compilation error with BE_USE_PERF_COUNTERS=0) looks unrelated to the preprocessing issue.
Issue for the preprocessing issue: #208

s-hadinger, 2021-12-15:
I will fix this

s-hadinger, 2021-12-15:
See #209

closed 2021-12-15 by s-hadinger