berry-lang/berry #208: pycoc: berry_conf.h is not properly parsed

paulrouget, 2021-12-15:
If I add `#includes` if `#ifndef` in the header, pycoc skips all of those instructions.

The reason why I add include and ifndef is to have separate files for different configurations.

paulrouget, 2021-12-15:
If we can't use regular preprocessing instructions in that file, maybe a giant warning should be added to the file, it can lead to inconsistencies.

s-hadinger, 2021-12-15:
I agree that define parsing in coc/pycoc is very basic and based on pattern matching. It would be an overkill to first do a full preprocessor parsing, unless you know an easy way to do it.

Actually you can statically enable all modules you will need in the union of builds, and not include the references to the modules/classes in your C code with proper ifdefs. The linker will discard what you don't need.

The main drawback of this, is that the global map of constant strings might contain string literals that you don't need. This may be a minor or major issue.

s-hadinger, 2022-07-26:
Closing since it will not improve

closed 2022-07-26 by s-hadinger