berry-lang/berry #273: fix CI failed

Skiars, 2022-07-07:
Seems to be due to this commit: 900e0030d14cddd4f58d3a572bd5d1910f01dccb

An ASAN error was triggered:
```
==2684==ERROR: AddressSanitizer: global-buffer-overflow on address 0x564558c9c421 at pc 0x564558bbee6b bp 0x7ffe6b71a6a0 sp 0x7ffe6b71a690
READ of size 1 at 0x564[55](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:56)8c9c421 thread T0
    #0 0x[56](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:57)4558bbee6a in next src/be_lexer.c:109
    #1 0x564558bc30bd in scan_string src/be_lexer.c:422
    #2 0x564558bc3ebc in lexer_next src/be_lexer.c:546
    #3 0x564558bc5565 in be_lexer_scan_next src/be_lexer.c:608
    #4 0x564558c75842 in be_parser_source src/be_parser.c:1734
    #5 0x564558c1[57](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:58)55 in m_parser src/be_exec.c:159
    #6 0x5645[58](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:59)c140[60](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:61) in be_execprotected src/be_exec.c:117
    #7 0x564558c15bac in be_protectedparser src/be_exec.c:176
    #8 0x564558c1[62](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:63)83 in be_loadbuffer src/be_exec.c:210
    #9 0x564558b8f078 in m_compile_str src/be_baselib.c:420
    #10 0x564558b8f435 in l_compile src/be_baselib.c:454
    #11 0x564558c0cee6 in vm_exec src/be_vm.c:1141
    #12 0x564558c0f8a6 in do_closure src/be_vm.c:1237
    #13 0x564558c10cc0 in be_dofunc src/be_vm.c:1273
    #14 0x564558c17301 in m_pcall src/be_exec.c:300
    #15 0x564558c14060 in be_execprotected src/be_exec.c:117
    #16 0x564558c174ac in be_protectedcall src/be_exec.c:312
    #17 0x564558ba8e96 in be_pcall src/be_api.c:1013
    #18 0x564558c813d8 in doscript default/berry.c:207
    #19 0x564558c81568 in load_script default/berry.c:223
    #20 0x564558c826c1 in analysis_args default/berry.c:337
    #21 0x564558c82978 in main default/berry.c:365
    #22 0x7f7e894fe082 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #23 0x564558b6f09d in _start (/home/runner/work/berry/berry/berry+0x17809d)
0x564558c9c421 is located [63](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:64) bytes to the left of global variable '*.LC82' defined in 'src/be_lexer.c' (0x5[64](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:65)558c9c460) of size 48
  '*.LC82' is ascii string '/home/runner/work/berry/berry/src/be_lexer.gcda'
0x564558c9c421 is located 0 bytes to the right of global variable 'eos' defined in 'src/be_lexer.c:104:27' (0x564558c9c420) of size 1
  'eos' is ascii string ''
SUMMARY: AddressSanitizer: global-buffer-overflow src/be_lexer.c:109 in next
Shadow bytes around the buggy address:
  0x0ac92b18b830: f9 f9 f9 f9 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b840: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b850: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b860: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b870: 00 00 00 00 03 f9 f9 f9 f9 f9 f9 f9 03 f9 f9 f9
=>0x0ac92b18b880: f9 f9 f9 f9[01]f9 f9 f9 f9 f9 f9 f9 00 00 00 00
  0x0ac92b18b890: 00 00 f9 f9 f9 f9 f9 f9 00 00 00 00 00 00 00 00
  0x0ac92b18b8a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b8b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b8c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac92b18b8d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==2[68](https://github.com/berry-lang/berry/runs/7226310109?check_suite_focus=true#step:4:69)4==ABORTING
```

s-hadinger, 2022-07-07:
I will take a look. Good catch

s-hadinger, 2022-07-26:
Closing since fixed

closed 2022-07-26 by s-hadinger