berry-lang/berry #281: Publishing the VSC tooling on marketplace

feelfreelinux, 2022-07-31:
Hi! Thanks a lot for the work on this project - I use it as a scripting engine in my project, and it's doing it's job absolutely wonderfully. 

Currently, the VSC syntax highlighter is not available from the VSC market. Is it okay if I release it?

toaster-code, 2022-07-31:
+1 to this.

I am trying to understand how to add the plugin to my vscode. @feelfreelinux can you explain how to do it?

feelfreelinux, 2022-07-31:
> +1 to this.
> 
> I am trying to understand how to add the plugin to my vscode. @feelfreelinux can you explain how to do it?

In order to do this, go to the tools/... folder, build the plugin as .vsix and then in your vscode, execute "Install from VSIX" from the command palatte. This should open a file picker where you can navigate to your built plugin.

Skiars, 2022-08-01:
All code in this project is distributed under the MIT license, so you can use it for almost any purpose.

Also, I would very much welcome this.

feelfreelinux, 2022-08-01:
Great! In that case, I've packaged and published [the tooling.](https://marketplace.visualstudio.com/items?itemName=berry.berry). In the future, I'll try to expand the extension a little bit.

closed 2022-08-01 by feelfreelinux