berry-lang/berry #304: Warning in m_setbytes

hiperiondev, 2022-10-27:
[Compile] src/be_byteslib.c
src/be_byteslib.c: En la función ‘m_setbytes’:
src/be_byteslib.c:971:25: aviso: comparison of integer expressions of different signedness: ‘long unsigned int’ and ‘int32_t’ {también conocido como ‘int’} [-Wsign-compare]
  971 |         if ((size_t)idx >= attr.len) { idx = attr.len; }
      |                         ^~
src/be_byteslib.c:984:26: aviso: comparison of integer expressions of different signedness: ‘int32_t’ {también conocido como ‘int’} and ‘size_t’ {también conocido como ‘long unsigned int’} [-Wsign-compare]
  984 |             if (from_len >= from_len_total) { from_len = from_len_total; }
      |                          ^~
src/be_byteslib.c:986:45: aviso: comparison of integer expressions of different signedness: ‘long unsigned int’ and ‘int32_t’ {también conocido como ‘int’} [-Wsign-compare]
  986 |         if ((size_t) idx + (size_t)from_len >= attr.len) { from_len = attr.len - idx; }
      |                                             ^~

s-hadinger, 2022-10-27:
Thanks. Yes I'm aware. I will fix it in a couple of days.

hiperiondev, 2022-12-14:
Some notice about this?

closed 2025-11-03 by s-hadinger