berry-lang/berry #309: How to use with arduino IDE?

dragoncoder047, 2022-11-13:
I figured compiling Berry in the Arduino IDE would be simple-- copy over some files, hit upload, and bam.

So i `git clone`d Berry next to a black arduino sketch.

Directory structure looked like this:

```
mysketch/
    mysketch.ino
    berry/
        ...
```

And `mysketch.ino`:

```cpp
extern "C" {
#include "berry/src/berry.h"
}
void setup() {}
void loop() {}
```
I tried to compile, but the compiler could not find `berry.h`. So i copied all the files in `berry/src` next to the sketch (not in a subfolder) but then it couldn't find `berry_conf.h`.

How do I need to structure my arduino folders to use this in a sketch?

I cannot use PlatformIO because there is a bug on my OS.

s-hadinger, 2022-11-13:
Sorry, you're alone here. Platform is the way to go

dragoncoder047, 2022-11-13:
Is there some tool that would concatenate all the needed source files into one big `berry_all.h`/`berry_all.c` that could be included?

Skiars, 2022-11-13:
I haven't used arduino, but from my experience with Visual Studio, it should be sufficient to add all the source files to the project.

It is recommended to look at how the Makefile contains these source files. However, that the `src` directory is Berry's core code, while the `default` directory is the implementation of the default interpreter program (aka berry.exe). You will need to refer to `default` to implement your own interpreter, even if it is simply an evaluation of strings.

dragoncoder047, 2022-11-13:
> I haven't used arduino, but from my experience with Visual Studio, it should be sufficient to add all the source files to the project.

I though that would be sufficient too, but unfortunately Arduino is super weird. It copies files to a temp directory before it compiles, and it only copies the files that are right next to the source `.ino` file. If I have to try PlatformIO again, I will, and fingers crossed it was a cosmic ray last time and it will work.

If it doesn't work, either I will have to look into sidestepping the Arduino IDE manually and compiling/flashing via command line, or I have [a plan B that I will need to finish](https://github.com/dragoncoder047/tehssl).

closed 2022-11-13 by dragoncoder047

rjjrbatarao, 2022-12-28:
I got it working on arduino ide, took me a month with gotchas and finally compiled, my only problem now is how to reuse esp classes and expose them to berry.   https://github.com/rjjrbatarao/berry_arduino