berry-lang/berry #312: Documentation on exposing c classes to berry as class

rjjrbatarao, 2022-12-28:
Are there any documentations on how to expose c class as berry class? I'm still confused how to do this. Im trying to expose esp8266 library classes to be used in berry as a class. Thank you for this great library.

s-hadinger, 2022-12-28:
The easiest way if it is not performance sensitive is to use https://github.com/berry-lang/berry_mapping

Otherwise you need to do it the hard way. 

There are plenty of examples in Tasmota https://github.com/arendst/Tasmota/tree/development/lib/libesp32/berry_tasmota/src

darcyg, 2023-01-01:
@s-hadinger 
I hope to extend the function of tasmota's berry by adding a custom xxx.be in embedded. but encountered a problem
I can't generate solidified_xxx_xxx.h in solidify.
I tried to compile the berry of this project into bin format, and then put it under /usr/local/bin/
But I can't generate solidified_xxx_xxx.h file with `berry -s -g` or `cat solidify_all.be | berry -s -g`.
May I ask how to generate solidified_xxx_xxx.h.
thanks

s-hadinger, 2023-01-02:
You just need to compile a local version of Berry. Use `make` in `lib/libesp32/berry`. Then change directory to `lib/libesp32/berry_tasmota` and use `./solidify_all.be`

darcyg, 2023-01-02:
> You just need to compile a local version of Berry. Use `make` in `lib/libesp32/berry`. Then change directory to `lib/libesp32/berry_tasmota` and use `./solidify_all.be`

After I execute it, I get an error

> error: cannot open file '-s -g'.

darcyg, 2023-01-02:
@s-hadinger 
thanks
I have successfully executed with cat solidify_all.be | ../berry/berry -s -g and obtained the solidifed_xxx.h file
But I found that the four files i2c_axp192.be / i2c_axp202.be / lv_tasmota_widgets.be / lv_tasmota.be cannot be compiled
For the two files of i2c, it is prompted that the I2C_Driver class cannot be found during compilation. I couldn't get both files to compile after i2c_driver.be, had to remove
The two files of lv cannot be compiled. an error message

> Parsing:  python_compat.be
Parsing:  zigbee_zcl_frame.be
Parsing:  energy.be
Parsing:  test_crypto.be
Parsing:  animate_module.be
Parsing:  trigger_class.be
Parsing:  lv_tasmotaa_widgets.be
syntax_error: string:96: '_lvgl' undeclared (first use in this function)
stack traceback:
	<native>: in native function
	<unknown source>: in function `parse_file`
	<unknown source>: in function `main`

Another error showing

>Parsing:  python_compat.be
Parsing:  zigbee_zcl_frame.be
Parsing:  energy.be
Parsing:  test_crypto.be
Parsing:  animate_module.be
Parsing:  trigger_class.be
Parsing:  dyn.be
Parsing:  persist.be
Parsing:  zigbee_zb_coord.be
Parsing:  lv_tasmota.be
syntax_error: string:29: 'lv_clock' undeclared (first use in this function)
stack traceback:
	<native>: in native function
	<unknown source>: in function `parse_file`
	<unknown source>: in function `main`

closed 2024-05-14 by s-hadinger