berry-lang/berry #318: Embedding berry on an MCU, where to start?

hello-world-dot-c, 2023-03-24:
I want to use berry on an ARM Cortex device that is programmed bare-metal or using a lightweight cooperative-scheduling OS. This means I need to be able to run scripts in the interpreter without relinquishing control completely. Can I do this with berry? And where should I look for information on how to embed berry in my application? There doesn't seem to be any documentation or examples for this type of use.

s-hadinger, 2023-03-25:
You can start looking here https://github.com/berry-lang/berry/wiki/Appendix-C

Keep in mind that Berry does not support multi-threading, and is not reentrant. You can let Berry give back control from time to time but you should not call back Berry in the middle of an current execution.

Alternatively, you can have multiple Berry vm coexisting, but this may consume too much memory.

hello-world-dot-c, 2023-03-25:
> You can start looking here https://github.com/berry-lang/berry/wiki/Appendix-C

Thank you, have checked it out and will try. From the looks of it, there doesn't seem to be an option to disable all floating-point support, is there?
> You can let Berry give back control from time to time 

This is interesting. How do I do that?
> Alternatively, you can have multiple Berry vm coexisting, but this may consume too much memory.

Agreed, likely not an option.

s-hadinger, 2023-03-25:
> Thank you, have checked it out and will try. From the looks of it, there doesn't seem to be an option to disable all floating-point support, is there?

No there isn't, it would require some work to do.

> > You can let Berry give back control from time to time
> 
> This is interesting. How do I do that?

You can set the `BE_VM_OBSERVABILITY_SAMPLING` to have a hook called every 2^n instructions. It allows you to regain control periodically, and potentially kill infinite loops.

https://github.com/berry-lang/berry/blob/5084b48142b0482ed57d09f0e88677df2e53d015/default/berry_conf.h#L77-L84

hello-world-dot-c, 2023-04-12:
> There is already a port to ESP32 that I made a while ago. Maybe it can help you:
> 
> https://github.com/hiperiondev/esp32-berry-lang

Thank you that would be helpful but I got stuck somewhere else. For various reasons that aren't relevant here I would have to also set up a simulation of the embedded system using a Windows DLL built under VS and when I tried integrating berry it turned out that the MS compiler can't build it. It could be because it has support up to C17 but still doesn't completely implement C99, I don't know. But it throws errors around berry's use of tokenized macros, run-time declaration of arrays, etc.

shreeve, 2023-12-01:
@hiperiondev - How hard would it be for you to create a little uf2 or other firmware image with the latest version of BerryScript? I think having a little platform like this with BerryScript would be amazing!

Take a look at https://kalumajs.org for a Javascript based similar project for Raspberry Pi Pico (and Pico W) boaed.

closed 2024-04-17 by s-hadinger