berry-lang/berry #338: Strange GC behaviour

AlixANNERAUD, 2023-05-14:
Hello,
With berry mapping, I have this function that create an instance of the 'Instruction_Type' class, here is the simplified version :
```cpp
void *Berry_This_Get_Instruction(bvm *V)
{
    void* Pointer = be_malloc(V, sizeof(Instruction_Class));
    Log_Verbose("Allocate instruction : %p ", Pointer);
    return Pointer;
}
BE_FUNC_CTYPE_DECLARE(Berry_This_Get_Instruction, "Instruction_Type", "@");
// Other stuff
/*
@const_object_info_begin
module This (scope:global)
{
    Get_Instruction, ctype_func(Berry_This_Get_Instruction)
    ...
}
@const_object_info_end
*/
```
Since this function is called multiple times, multiple instance are created.
Since instance `Instruction_Type` are "short lived" variables, it's used an then trown away by the function.
When the GC trigger, it try to dealocate everything twice, here is the serial output of the ESP32 :
```
Allocate instruction : 0x3ffdcc90
Allocate instruction : 0x3ffdccd0 
Allocate instruction : 0x3ffdcd10
... (Multiple allocation)
... (Multiple deallocation)
De-allocate instruction : 0x3ffdcc90
De-allocate instruction : 0x3ffdccd0
De-allocate instruction : 0x3ffdcc90
Throw exception : 2
... (Multiple deallocation)
De-allocate instruction : 0x3ffdcc90
De-allocate instruction : 0x3ffdccd0
De-allocate instruction : 0x3ffdcc90
Throw exception : 2

(Abort called by Berry VM)
```
As you may notice, the GC seems to try to deallocate twice the same memory. And also it throw the berror 2 which correspond to malloc_failed. It is not an issue with regular malloc (~150k free mem when run).
Here is the code responsible for the `Instruction_Type` deallocation / deinit : 
void Berry_Instruction_Class_Deinitialize(bvm* V, Instruction_Class *Instruction)
{
    Log_Verbose("De-allocate instruction : %p", Instruction);
    be_free(V, Instruction, sizeof(Instruction_Class));
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Deinitialize, "", "@.");

And here is the `Instruction_Type` berry mapping declaration (it's registered in `be_modtab.c`)
```cpp
/* @const_object_info_begin
class be_class_Berry_Instruction_Type(scope:global, name:Instruction_Type)
{
    _p, var
    init, ctype_func(Berry_Instruction_Class_Initialize)
    deinit, ctype_func(Berry_Instruction_Class_Deinitialize)
    ...
}
@const_object_info_end */
```
The Berry VM / Interpreter is run on the ESP32, everything seemed to work fine so far. Am I doing something wrong ?

s-hadinger, 2023-05-14:
Can you share the function `Berry_Instruction_Class_Initialize`?

AlixANNERAUD, 2023-05-14:
Yes, here it is :
```cpp
void *Berry_Instruction_Class_Initialize(bvm* V)
{
    void* Pointer = be_malloc(V, sizeof(Instruction_Class));
 
    Log_Verbose("Initialize instruction : %p", Pointer);
    
    return Pointer;
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Initialize, "+_p", "@");
```
What does this have to do with anything? Is this function not supposed to intervene? Because the object is already instantiated by `Berry_This_Get_Instruction`

s-hadinger, 2023-05-14:
I wanted to check how you did the allocation as well. I can't reproduce the problem, I'm doing exactly the same scheme with `int64` and it works flawleslly.

Can you also share the full stacktrace? What's calling be_realloc is what's most interesting. It can't be `be_free()` because in such case this portion of code is not reachable.

AlixANNERAUD, 2023-05-14:
Well, here is the full file Instruction_Type file : 
```cpp
#include "Software/Berry/Berry.hpp"
extern "C"
{
#include "be_constobj.h"
#include "be_mapping.h"
}

using namespace Xila_Namespace;

void *Berry_Instruction_Class_Initialize(bvm* V, Module_Class *Sender = NULL, Module_Class *Receiver = NULL)
{
    void* Pointer = be_malloc(V, sizeof(Instruction_Class));
    if ((Sender == NULL) && (Receiver == NULL))
        Pointer = new (Pointer) Instruction_Class();
    else
        Pointer = new (Pointer) Instruction_Class(Sender, Receiver);

    Log_Verbose("Instruction", "Initialize instruction : %p / %p / %p", Pointer, Sender, Receiver);
    
    return Pointer;
}

BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Initialize, "+_p", "@[..");

void Berry_Instruction_Class_Deinitialize(bvm* V, Instruction_Class *Instruction)
{
    Log_Verbose("Instruction", "Deinitialize : %p", Instruction);
    be_free(V, Instruction, sizeof(Instruction_Class));
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Deinitialize, "", "@.");

void Berry_Instruction_Class_Set_Sender(Instruction_Class *Instruction, Module_Class *Sender)
{
    Instruction->Set_Sender(Sender);
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Set_Sender, "", "..");

void Berry_Instruction_Class_Set_Receiver(Instruction_Class *Instruction, Module_Class *Receiver)
{
    Instruction->Set_Receiver(Receiver);
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Set_Receiver, "", "..");

void *Berry_Instruction_Class_Get_Sender(Instruction_Class *Instruction)
{
    return Instruction->Get_Sender();
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Get_Sender, "c", ".");

void *Berry_Instruction_Class_Get_Receiver(Instruction_Class *Instruction)
{
    return Instruction->Get_Receiver();
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Get_Receiver, "c", ".");

int Berry_Instruction_Class_Graphics_Get_Code(Instruction_Class *Instruction)
{
    return (int)Instruction->Graphics.Get_Code();
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Graphics_Get_Code, "i", ".");

void *Berry_Instruction_Class_Graphics_Get_Target(bvm* V, Instruction_Class *Instruction)
{
    Graphics_Types::Object_Type *Pointer = (Graphics_Types::Object_Type *)be_malloc(V, sizeof(Graphics_Types::Object_Type));
    return new (Pointer) Graphics_Types::Object_Type(Instruction->Graphics.Get_Target());
}
BE_FUNC_CTYPE_DECLARE(Berry_Instruction_Class_Graphics_Get_Target, "Graphics.Object_Type", "@.");

[... 383 lines cut ...]
| 181839 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1e2c
| 181850 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1dec
| 181861 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1dac
| 181872 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1d6c
| 181883 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1d2c
| 181894 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1cec
| 181905 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe1cac
| 181916 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe687c
| 181927 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe683c
| 181939 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe67fc
| 181950 ms | Instruction.Berry_Instruction_Class_Deinitialize() | (Instruction.cpp:27) |  Verbose | Deinitialize : 0x3ffe67bc
Throw exception: 2

abort() was called at PC 0x4018ab13 on core 1

Backtrace: 0x400844e9:0x3ffe6340 0x4008fd05:0x3ffe6360 0x400961e9:0x3ffe6380 0x4018ab13:0x3ffe6400 0x4018c646:0x3ffe6420 0x401cc80a:0x3ffe6440 0x4019a0af:0x3ffe6460 0x4019a236:0x3ffe64c0 0x401911f0:0x3ffe64e0 0x40191449:0x3ffe6500 0x4018b635:0x3ffe6520 0x4018b931:0x3ffe6540 0x401913d9:0x3ffe6560 0x400dac38:0x3ffe6580 0x400ddefd:0x3ffe65a0 0x400ddf45:0x3ffe65d0 0x400de2f5:0x3ffe65f0 0x400fb34d:0x3ffe6640

  #0  0x400844e9:0x3ffe6340 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008fd05:0x3ffe6360 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x400961e9:0x3ffe6380 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4018ab13:0x3ffe6400 in be_throw at lib/berry/src/be_exec.c:87
  #4  0x4018c646:0x3ffe6420 in be_realloc at lib/berry/src/be_mem.c:108
  #5  0x401cc80a:0x3ffe6440 in Berry_Object_Class_Deinitialize_607AD9B3_9A5E_4076_89DD_81ED8038C914(bvm*, Xila_Namespace::Graphics_Types::Object_Class*) at lib/Xila/src/Software/Berry/Interpreter/Generator/Generated/Graphics.cpp:33
  #6  0x4019a0af:0x3ffe6460 in be_call_c_func at lib/berry_mapping/src/be_class_wrapper.c:464 (discriminator 1)
```

s-hadinger, 2023-05-14:
Thanks. The stacktrace shows that the crash appears in `Berry_Object_Class_Deinitialize`. You haven't included the source code for this one.

Please bear in mind that `deinit()` is a last chance and very constraint: you cannot allocate memory, and you cannot throw an exception.

What appears here is that an exception is thrown during GC, which aborts the last phase of cleaning. This is why a new GC is triggered right after, and wrongly deinits objects a second time.

s-hadinger, 2023-05-14:
Hmmm. I have never tried `deinit` with Berry inheritance. But indeed, inheritance is implemented with multiple instances, so I suppose that each level of inheritance receives a call to `deinit()` if such function is defined.

I still don't see how `be_realloc at lib/berry/src/be_mem.c:108` could be reached by a call to `be_free()`. In such call `new_size` is equal to `0` and the code stops at "Case 2: deallocate".

s-hadinger, 2023-05-14:
Thinking again, the call to `be_free()` should be done at the same level of sub-class that when it was allocated. Doing `be_free()` at multiple levels would be wrong.

AlixANNERAUD, 2023-05-14:
I think so ...

AlixANNERAUD, 2023-05-14:
I just checked, adding an if statement solves the problem. However, I am surprised that this is not supported directly in Berry?

s-hadinger, 2023-05-14:
You mean calling `be_free()` on a null pointer? I can indeed add a test.

closed 2023-05-15 by s-hadinger