berry-lang/berry #340: Constructor that take ptr as argument never called

AlixANNERAUD, 2023-05-19:
Here is the following berry code :
```
Foo = Get_Ptr()
print(Foo) # print : <ptr: 0x3fca0a64> for example

Bar = Test(Foo) #Class constructor taking pointer as argument
Bar = Test(15)
```

The constructor is declared as CTYPE_FUNC with berry mapping as follows : 

```cpp
class Test_Class()
{
// Stuf here
};

void* Test_Init(bvm* VM, void* PTR)
{
Log_Format("Test_Init with ptr : %p", PTR);
return be_malloc(VM, sizeof(Test_Class));
}
BE_FUNC_CTYPE_DECLARE(Test_Init, "+_p", "@.")

// Berry mapping @const_object_info_begin declaration of the class and module
```

Once the code is ran, `Bar = Test(Foo)` is never called,  but `Bar = Test(15)` is (of course invalid), but called.
I don't understand why.

s-hadinger, 2023-05-19:
Oh, I realize this is a leftover from early LVGL mapping. This is not documented, but currently when you pass a `comptr` to a constructor, it is directly assigned to the variable and the function is not called.

I think I should change this to not call the function only if the pointer to the function is NULL. Let me check.

AlixANNERAUD, 2023-05-19:
That would be great ! Thank you for your quick answer.

AlixANNERAUD, 2023-05-21:
Update ?

s-hadinger, 2023-05-21:
Sorry not yet. I tried a quick fix but it failed

s-hadinger, 2023-05-22:
Can you elaborate more about what you need? Can you make sure you don't send a comptr as first argument?

AlixANNERAUD, 2023-05-22:
Well, since in my case the `comptr` point to an object allocated outside Berry, it cannot be passed as a class instance to berry because the GC will clean it. Unless there is another way to practice, I don't see how else. However I totally understand the mechanism, and yes indeed, it makes sense.

s-hadinger, 2023-05-22:
My point is that it would work if you pass a dummy first argument, and the comptr as a second argument.

AlixANNERAUD, 2023-05-22:
Ok, well, thank you, I will try to do it like that.

s-hadinger, 2023-05-22:
If you allocate memory outside of Berry but still want to benefit from the GC, you can use `comobj`. `comobj` are managed by GC and you register a function that will be called when memory is deallocated.

AlixANNERAUD, 2023-05-22:
Comobj ? Do you have the link to its documentation  ?

s-hadinger, 2023-05-22:
The way I do in Tasmota:

```C

/* generic destroy method for comobj, just call be_os_free() on the pointer */
int be_commonobj_destroy_generic(bvm* vm)
{
    int argc = be_top(vm);
    if (argc > 0) {
        void * obj = be_tocomptr(vm, 1);
        if (obj != NULL) { be_os_free(obj); }
    }
    be_return_nil(vm);
}
```

And then I can allocate an object. Example from Tasmota crypto:
```C
[...]
    // Initialize an AES CTR structure with the secret key
    br_aes_small_ctr_keys * ctr_ctx = (br_aes_small_ctr_keys *) be_os_malloc(sizeof(br_aes_small_ctr_keys));
    if (!ctr_ctx) { be_throw(vm, BE_MALLOC_FAIL); }
[...]
    be_newcomobj(vm, ctr_ctx, &be_commonobj_destroy_generic);
    be_setmember(vm, 1, ".p1");
[...]
```

s-hadinger, 2023-05-22:
> Comobj ? Do you have the link to its documentation ?

Unfortunately documentation of `comobj` is missing. I inferred it from the source.

AlixANNERAUD, 2023-05-22:
Thank you, however, I don't think this is suitable for my use. I especially don't want my object, allocated outside of berry, to be deallocated as it is also used outside of the VM.

s-hadinger, 2023-05-22:
Ok, got it. Then `comptr` is the right way.

AlixANNERAUD, 2023-05-22:
Thank you, though, this will be very useful

AlixANNERAUD, 2023-05-23:
I tried to pass an instance of an object as the first argument parameter, and I get the same behaviour.

closed 2024-10-12 by AlixANNERAUD