berry-lang/berry #346: f-strings for Berry

s-hadinger, 2023-06-20:
Now that I have some decent experience with programming in Berry, I'm am not completely happy with string formatting. `string.format()` is slightly verbose. I would like to add an equivalent to Python's f-strings.

Example:
`string.format("id=%i cluster=0x%04X name=%s", id, cluster, self.name)`

would become:
`f"id={id:i} cluster={cluster:04X} name={self.name}"`

I made a POC that sounds promising. Changing the parser would be a daunting task. Instead, I use a "transpiler" approach. The lexer contains code to detect f-string syntax and rewrite the code to generate the string.format syntax.

The lexer transforms `f"id={id:i} cluster={cluster:04X} name={self.name}"` to `("id=%i cluster=0x%04X name=%s", id, cluster, self.name)` and then just feeds the normal parser.

sfromis, 2023-06-20:
Also see discussion on Discord.
https://discord.com/channels/479389167382691863/826169659811168276/1120818883713040476

s-hadinger, 2023-07-15:
Done: https://github.com/berry-lang/berry/pull/355

closed 2023-07-15 by s-hadinger

barbudor, 2023-11-11:
Hi @s-hadinger 
Is that working only as compile-time like in python ? which means that I can't use a string variable ?

I'm looking if there is a way to use format string like in python:
```Python
# fulltopic = "{prefix}/{topic}/{subtopic}"
fulltopic = fulltopic=string.replace(string.replace(tasmota.cmd("fulltopic")['FullTopic'],"%prefix%","{prefix}"),"%topic%","{topic}")+"{subtopic}"
# publish_topic = "cmnd/devname/power"
publish_topic = fulltopic.format(prefix="cmnd", topic="devname", subtopic="power")
# or
publish_topic = fulltopic.format(**{"prefix":"cmnd", "topic":"devname", "subtopic":"power"})
```

As you imagine the idea is to use named variable in the format string instead of positional values

Thanks

s-hadinger, 2023-11-11:
f-strings are purely compile time synctactic sugar to the `format()` function. You could however write a helper function that does something similar.

barbudor, 2023-11-11:
Yes, indeed I simply string replace:
```
fulltopic=tasmota.cmd("fulltopic")['FullTopic']

def make_topic(prefix, topic, subtopic)
  return string.replace(string.replace(fulltopic,"%prefix%",prefix),"%topic%",topic) + subtopic
end
```

sfromis, 2023-11-11:
If you really want, you *can* have non-constant f-strings expanded by going via `compile`, to get the syntactic sugar expanded, but that's not very efficient.
```
var prefix='stat' var topic='device' var leaf='what'
var fformat=/fst-> compile("return f'" + fst + "'")()
print(fformat('{topic}/{prefix}/{leaf}'))
```

s-hadinger, 2023-11-11:
I wouldn't recommend using compile with dynamic content. This is always dangerous ; like eval() in many languages