berry-lang/berry #367: Evaluate Profile-Guided Optimization (PGO) and LLVM BOLT

zamazan4ik, 2023-10-07:
Hi!

Recently I did many Profile-Guided Optimization (PGO) benchmarks on multiple projects (including interpreters and compilers like CPython, Rustc, Clang, GCC, and many others) - the results are available [here](https://github.com/zamazan4ik/awesome-pgo). So that's why I think it's worth trying to apply PGO to Berry tooling.

I can suggest the following things to do:
* Evaluate PGO's applicability to Berry tooling with benchmarks.
* If PGO helps to achieve better performance - add a note to Berry's documentation about that. In this case, users and maintainers will be aware of another optimization opportunity for Berry.
* Provide PGO integration into the build scripts. It can help users and maintainers easily apply PGO for their own workloads.

Here are some examples of how PGO is already integrated into other projects' build scripts:
* Python: 
  - CPython: [README](https://github.com/python/cpython#profile-guided-optimization)
  - Pyston: [README](https://github.com/pyston/pyston#building)
* Rustc: a CI [script](https://github.com/rust-lang/rust/blob/master/src/ci/stage-build.py) for the multi-stage build
* GCC:
  - Official [docs](https://gcc.gnu.org/install/build.html), section "Building with profile feedback" (even AutoFDO build is supported)
  - A [part](https://github.com/gcc-mirror/gcc/blob/4832767db7897be6fb5cbc44f079482c90cb95a6/configure#L7818) in a "wonderful" `configure` script 
* Clang: [Docs](https://llvm.org/docs/HowToBuildWithPGO.html) 
* Go: [Bash script](https://github.com/golang/go/blob/master/src/cmd/compile/profile.sh)
* V8: [Bazel flag](https://github.com/v8/v8/blob/main/BUILD.gn#L184)
* ChakraCore: [Scripts](https://github.com/chakra-core/ChakraCore/tree/master/Build/scripts/pgo)
* Chromium: [Script](https://chromium.googlesource.com/chromium/src/build/config/+/refs/heads/main/compiler/pgo/BUILD.gn)
* Firefox: [Docs](https://firefox-source-docs.mozilla.org/build/buildsystem/pgo.html)
   - Thunderbird has PGO support too
* PHP - [Makefile command](https://github.com/php/php-src/blob/master/build/Makefile.global#L138) and old Centminmod [scripts](https://github.com/centminmod/php_pgo_training_scripts)
* MySQL: [CMake script](https://github.com/mysql/mysql-server/blob/8.0/cmake/fprofile.cmake)
* YugabyteDB: [GitHub commit](https://github.com/yugabyte/yugabyte-db/commit/34cb791ed9d3d5f8ae9a9b9e9181a46485e1981d)
* FoundationDB: [Script](https://github.com/apple/foundationdb/blob/1a6114a66f3de508c0cf0a45f72f3687ba05750c/contrib/generate_profile.sh)
* Zstd: [Makefile](https://github.com/facebook/zstd/blob/dev/programs/Makefile#L232)
* [Foot](https://codeberg.org/dnkl/foot): [Scripts](https://codeberg.org/dnkl/foot/src/branch/master/pgo)
* Windows Terminal: [GitHub PR](https://github.com/microsoft/terminal/pull/10071)
* Pydantic-core: [GitHub PR](https://github.com/pydantic/pydantic-core/pull/741)

After PGO, I can suggest evaluating [LLVM BOLT](https://github.com/llvm/llvm-project/blob/main/bolt/README.md) as an additional optimization step after PGO.

skiars, 2023-10-09:
This should be able to be added once CMake build scripts #370 are supported? I'm not going to do this in Makefile because CMake is better.

zamazan4ik, 2023-10-09:
> This should be able to be added once CMake build scripts https://github.com/berry-lang/berry/issues/370 are supported? I'm not going to do this in Makefile because CMake is better.

If you are talking about integrating PGO build mode directly to the build scripts - yeah, it could be integrated directly into the CMake scripts. As an example, you can look at the following projects:
* Clang: [CMake cache](https://github.com/llvm/llvm-project/blob/main/clang/cmake/caches/BOLT-PGO.cmake)
* MySQL: [CMake script](https://github.com/mysql/mysql-server/blob/8.0/cmake/fprofile.cmake)

However, you can add PGO support even without CMake (e.g. via a dedicated Makefile command) - it's all up to you.

skiars, 2023-10-09:
I want to do it in CMake because it's cross-platform and it can generate Makefiles.