berry-lang/berry #371: Standalone && and || trigger infinite loop

Lydxn, 2023-10-09:
Either of these two lines will trigger an infinite loop, where you would expect them to do nothing:

```berry
0 && 1
1 || 0
```

This behavior seems to magically resolve when the expression is assigned to a variable, e.g.

```berry
_ = 0 && 1
```

s-hadinger, 2023-10-10:
I cannot reproduce:

```
> 0 && 1
false
> 1 || 0
true
> def f() 0 && 1 end
syntax_error: stdin:1: strict: expression without side effect detected
> def f() return 0 && 1 end
> f()
false
```

skiars, 2023-10-10:
These lines need to be parsed together. Try `0 && 1 1 || 0` in REPL.

s-hadinger, 2023-10-10:
Ah yes indeed. What's interesting is when running in strict mode, it gets the following error (which is expected):
```
> 0 && 1 1 || 0
syntax_error: stdin:1: strict: expression without side effect detected
```

Now looking at the produced code:
```
> def f() 0 && 1 1 || 0 end
> import debug
> debug.codedump(f)
source 'stdin', function 'f':
; line 1
  0000  LDCONST	R0	K0
  0001  JMPF	R0	#0001
  0002  LDCONST	R0	K1
  0003  JMPT	R0	#0003
  0004  RET	0
```

There is definitely a problem at line 3.

JayXon, 2025-11-14:
here's a normal looking code that reproduces this issue
```berry
for i:1..100
 i%3 || print(i)
end
```

s-hadinger, 2025-11-14:
Thanks. I'm not sure I would hire somebody writing such code ;)
But I see the bug. Be aware it's a low priority bug.

s-hadinger, 2025-11-19:
Wow, good catch. Actually it is a completely different bug, since `for i:1..100 (i) end` gives the same error. It looks to be linked that it is enclosed in parenthesis.
I will look at it.